zephyrproject-rtos/zephyr #6912: OS X: setup instructions are wrong

andrewboie, 2018-04-03:
`brew install cmake ninja dfu-util doxygen qemu dtc python3 gperf
curl -O 'https://bootstrap.pypa.io/get-pip.py'
./get-pip.py
rm get-pip.py
cd ~/zephyr   # or to the folder where you cloned the zephyr repo
pip3 install --user -r scripts/requirements.txt`

The ./get-pip.py line does not work, the script is not executable and requires sudo anyway.

andrewboie, 2018-04-04:
Crosstool setup has this command, which errors out
`$ brew tap homebrew/dupes
Error: homebrew/dupes was deprecated. This tap is now empty as all its formulae were migrated.`

andrewboie, 2018-04-04:
Although it used to be broken, ct-ng in brew works well now in High Sierra. I think we should delete the section about building from source.

andrewboie, 2018-04-04:
We should *definitely* delete the section on "Verifying the Configuration of the Toolchain".
The config files checked into the tree should be in a state where they Just Work if the recommended paths are used.

andrewboie, 2018-04-04:
The config files checked into the tree do NOT have the changes in "Verifying the Configuration of the Toolchain" made.
Why is this left as an exercise for the user??

```
...
CT_LOCAL_TARBALLS_DIR="/Volumes/CrossToolNG/src"
# CT_SAVE_TARBALLS is not set
CT_WORK_DIR="${CT_TOP_DIR}/.build"
CT_PREFIX_DIR="/Volumes/CrossToolNG/x-tools/${CT_TARGET}"
CT_INSTALL_DIR="${CT_PREFIX_DIR}"
# Following options prevent link errors
CT_WANTS_STATIC_LINK=n
CT_CC_STATIC_LIBSTDCXX=n
...
```

andrewboie, 2018-04-04:
OK I'm dead in the water trying to build the toolchain.
```
[EXTRA]  Preparing working directories
gstat: cannot read file system information for '%Lp': No such file or directory
```

andrewboie, 2018-04-04:
It seems the logic in ct-ng to detect whether gstat works like BSD or GNU is broken.
I hacked it by changing

```
andrew@magrathea:/usr/local/lib/crosstool-ng-1.23.0
$ ag gstat
scripts/functions
579:                mode="$(gstat -c '%a' "$(dirname "${dir}")")"
582:                mode="$(gstat -f '%Lp' "$(dirname "${dir}")")"
```

Line 582 to just 'stat' which is the OS X BSD stat tool. Seems to work?

andrewboie, 2018-04-04:
Anas
3:02 oh
3:02 i know what the issue is
3:02 you need to run 'make oldconfig'
3:03 the configs you have are probably for Linux and have GNU in them
3:03 'make oldconfig; would reset to BSD
3:03 I remember now, hit this some time ago
3:04 CT_CONFIGURE_has_stat_flavor_GNU=y
3:05 should be CT_CONFIGURE_has_stat_flavor_BSD=y on the mac

andrewboie, 2018-04-04:
apboie
3:47 > CT_CONFIGURE_has_stat_flavor_BSD=y
3:47 not as simple as this.
3:48 It is trying to use gstat from brew coreutils as BSD when it is actually GNU
3:48 so it is the opposite.

andrewboie, 2018-04-04:
I'm now stuck on #6911

carlescufi, 2018-04-05:
There's a lot of stuff in here, let's start with `get-pip.py`?

carlescufi, 2018-04-05:
@andrewboie a first pass at fixing some of the issues you mention: https://github.com/zephyrproject-rtos/zephyr/pull/6949

carlescufi, 2018-05-02:
Looked at this again. Seems like the issue is with `ct-ng`, I have no idea how to solve that since I have never used it. @nashif can you take a look?

carlescufi, 2018-05-02:
@andrewboie actually you say that you are stuck on #6911 but that is closed now. Is this present issue also solved then?