zephyrproject-rtos/zephyr #6925: Provide Reviewer Guidelines as part of the documentation

nashif, 2018-04-04:
Provide Reviewer Guidelines as part of the documentation as discussed in the TSC.

erwango, 2019-07-02:
Seems that this ticket has been adressed since I've found this section in doc:
https://docs.zephyrproject.org/latest/development_process/dev_env_and_tools.html?highlight=checkpatch#reviewers-shall-not-request-changes-without-comments-or-justification

Though, I've question on the following:
```
[...]. Reviews will be dismissed following the criteria below:
The review is unrelated to the code change or asking for unjustified structural changes such as:

    Split the PR
    Split the commits
```

I'm not sure to agree requesting splitting the PR or the commit is an unjustified request.
Also it is in contradiction with:
https://docs.zephyrproject.org/latest/contribute/index.html#other-commit-expectations
`Each commit must address a single identifiable issue and must be logically self-contained. Unrelated changes should be submitted as separate commits.`

carlescufi, 2021-03-11:
This is already addressed as pointed out by @erwango .

closed 2021-03-11 by carlescufi