zephyrproject-rtos/zephyr #6958: [PWM] Update PWM API for dynamic prescaler computation

erwango, 2018-04-06:
As discussed in [#5469](https://github.com/zephyrproject-rtos/zephyr/pull/5469#issuecomment-373422463), current PWM API prevents dynamic prescaler computation.

Indeed, using get_cycles_per_sec in pin_set prevents to adjust PWM prescaler to the requested cycle duration and hence imposes to set it before compilation. Hence PWM application is limited to work in the duration range matching harcoded prescaler value.

I propose to remove get_cycles_per_sec from API and let driver compute it internally inside pin_set.
It seems it is not used today (not in PWM tests/samples at least).

henrikbrixandersen, 2021-01-19:
This needs rethinking after introduction of the PWM capture API functions, which also use the `pwm_get_cycles_per_sec()` API function.

manoj153, 2021-05-24:
Is there a detection mechanism to log error that the provided `period` is not supported based on the base pre-scaler set value?

pszkotak, 2022-01-26:
If I understand correctly with current implementation there is impossible to have e.g. LED blinking with higher period that 250ms using PWM based driver (as in https://github.com/zephyrproject-rtos/zephyr/tree/main/samples/basic/blinky_pwm).

It limits some applications or makes a need for weird workarounds.

@carlescufi, @anangl, is it planned for development? Do you have any estimations?

Tooniis, 2024-10-12:
I'm trying to generate audio tones with timer output compare on STM32 and the fixed prescaler means prevents me from generating accurate frequencies on 16-bit timers while also limiting the range of tone I can generate. Has there been any progress on this?

erwango, 2024-10-14:
> I'm trying to generate audio tones with timer output compare on STM32 and the fixed prescaler prevents me from generating accurate frequencies on 16-bit timers while also limiting the range of tones I can generate. Has there been any progress on this?

No.