zephyrproject-rtos/zephyr #6961: The 'app' library and additional application libraries created are linked differently

SebastianBoe, 2018-04-06:
Users should be able to describe complex applications that are organized into a hierarchy of libraries with different build properties.

To this end, it must be documented and exemplified how to create and use new app CMake libraries.

The desired model for application libraries is that they be organized as pure CMakeLists.txt libraries that could in theory be copied verbatim from a CMake project. So, no reliance on Zephyr CMake extensions or conventions. This is believed to lower the barrier of entry for application developers as they do not have to learn about Zephyr's CMake model and can instead use the standard CMake API to describe their application.

But it is not clear how to achieve this goal. One problem relates to linking. In the zephyr model, libraries are --whole-archive'd together to resolve unresolved references. But in the CMake model libraries should instead be explicitly linked together. There exists build script code that is aware of the 'app' library and ensures that it is whole-archived, but there is no mechanism to ensure that extra application libraries are linked in (and it is not clear if it desirable to use whole-archive on these libraries).

SebastianBoe, 2018-04-06:
Example of the problem in action:
https://github.com/SebastianBoe/zephyr/commit/fa499525dade0973f08769e03758ad1174de001a

The above does build because app_lib is not whole-archived, this again causes symbols in the 'kernel' library to be omitted.

A temporary workaround/solution (not sure if this is a clean solution yet) is to use the CMake model and explicitly link 'app_lib' with the 'kernel' library, like so:

https://github.com/zephyrproject-rtos/zephyr/commit/5b8619f63466f5bd93b9a8e2cb23113655103302

This ensures that the 'kernel' references that 'app_lib' is using are not dropped. But is there any way for the user to figure this out on his own?

SebastianBoe, 2018-04-06:
Issue originally reported in: https://github.com/zephyrproject-rtos/zephyr/issues/5554#issuecomment-377183960

@pboettch : Hi, any thoughts on this workaround? https://github.com/zephyrproject-rtos/zephyr/commit/5b8619f63466f5bd93b9a8e2cb23113655103302

pboettch, 2018-04-17:
Sorry for getting back a little late. I will try your work-around as soon as I can, but it looks promising. 

It seems (to be tested) that the `whole-archive` and `start/end-group`-stuff is not necessary when using [LINK_INTERFACE_MULTIPLICITY](https://cmake.org/cmake/help/v3.8/prop_tgt/LINK_INTERFACE_MULTIPLICITY.html#prop_tgt:LINK_INTERFACE_MULTIPLICITY) .

I tried playing around with it in zephyr's CMakeLists.txt (and getting rid of the not very cmakely overriding of linker-flags) but I must have missed something.

If this works, it would enable this features without further kludges to the CMake-build-system of zephyr.

SebastianBoe, 2018-04-17:
"It seems (to be tested) that the whole-archive and start/end-group-stuff is not necessary when using LINK_INTERFACE_MULTIPLICITY ."

When initially porting the build system from Kbuild to CMake I tried and failed to get rid of --whole-archive. I can't remember the details, but I remember using LINK_INTERFACE_MULTIPLICITY. So it is still an open question whether --whole-archive can be removed.

As an aside, if one eventually gets rid of --whole-archive one should make sure that the mechanism that replaces it is compatible with LTO.

pboettch, 2018-04-17:
With LTO you mean real LTO or garbage-collection via `gc-sections`?

SebastianBoe, 2018-04-17:
real LTO