zephyrproject-rtos/zephyr #6973: bad magic number in 'kconfiglib' error when generating docs

dbkinder, 2018-04-06:
When I try to generate documentation I'm getting an error:

    ~/zephyr$ make htmldocs
    Traceback (most recent call last):
      File "scripts/genrest/genrest.py", line 5, in <module>
        import kconfiglib
    ImportError: bad magic number in 'kconfiglib': b'\x03\xf3\r\n'
    Makefile:76: recipe for target 'kconfig' failed
    make[1]: *** [kconfig] Error 1
    Makefile:17: recipe for target 'htmldocs' failed
    make: *** [htmldocs] Error 2

dbkinder, 2018-04-06:
Hi @ulfalizer, not sure what's up here, can you help?

dbkinder, 2018-04-06:
After a bit of searching I found there was a ~/zephyr/doc/scripts/genrest/kconfiglib.pyc file.  No sure how it got created, but I deleted it and all's well now.

closed 2018-04-06 by dbkinder

ulfalizer, 2018-04-07:
Sorry, didn't see this until now.

Yeah, probably some kind of incompatibility from different Python versions or from Windows vs Linux or something. Can't recreate it. At least Python 2.7 and Python 3.6 seem to be able to deal with `.pyc`s from each other being there.

`.pyc` files are precompiled bytecode. They're generated when you `import` stuff, to save on parsing overhead later.