zephyrproject-rtos/zephyr #6974: arm_mpu: _get_region_attr() needs improvement

andrewboie, 2018-04-06:
This function doesn't bounds-check its arguments, the arguments don't exactly correspond to the offsets in the technical documentation (for example, the enable bit is included in the size parameter and the region parameters are pre-shifted) and hard-coded values for shift offsets are used instead of defines from CMSIS.

add __ASSERTS for bounds checks, use CMSIS for offsets/masks if possible

ioannisg, 2018-06-01:
Relates to #8018 
Lots of the suggested improvements should be done as part of extending the ARM_MPU driver to ARMv8-M

ioannisg, 2018-07-06:
@andrewboie this has been, mostly, addressed, I believe:
- CMSIS offsets are used now, instead of hard-coded values,
- the enable bit is not included in the size parameter. Instead, the enable bit is set by _region_init(.)
- all arguments, now, correspond to the respective offsets (including the AP)

The function still does notbound-check its arguments, but we can live with that, I think. The function is only called by helper functions that generate the proper set of MPU region attributes; if we trust the helper functions there is nothing we need to do for _get_region_attr(.)

S, as far as the ARM MPU is concerned, I think we can close this. I am not sure if this is still applicable for the NXP MPU. @MaureenHelm ? @agross-linaro? If it is, my proposal is to covert this to an NXP-specific issue.

ioannisg, 2019-02-05:
Obsolete, now, after #12817, for reasons stated above and also because the function does not exist.

closed 2019-02-05 by ioannisg