zephyrproject-rtos/zephyr #6991: Enhance test reporting and maintain one source for testcase meta data

nashif, 2018-04-09:
The plan to use a test management system (TMS) for maintaining our test cases requires an overhaul of our test-case meta-data and how we maintain them. The following is needed to allow for long-term reporting on a granular level:
- Need to have unique test case IDs
- Need to have a system for categorisation of the test-cases. At minimum we need 2 levels: section and sub-section. For example Kernel/Semaphore
- Need to track testcases on a fine grained level.  A testcase project in Zephyr might be testing too many thing, i.e different features and APIs in a sub-section, we will need to identify those tests and track them over time and we need to be able to trace back to APIs and sub-features.
- We need to know what tests we are expected to run without running them, this is needed to identify failures when a test is being executed where no PASS|FAIL can be identified. In suck cases where a test fails to run initially, the tests need to be marked as blocked.
- We want to maintain once source for all of the above information and avoid duplication between the test sources and the test management system.

For example, traditionally we had the semaphore API testcase with the following information:

```
tests:
  test:
    tags: kernel userspace
```
With the above requirement and changes in place, this will become:

```
common:
  cases:
      - sema_thread2thread
      - sema_thread2isr
      - sema_reset
      - sema_count_get
tests:
  kernel.semaphore:  # <--- <section>.<subsection>
    tags: kernel userspace
```

Every test of the four tests in this test application will have a unique ID:

- kernel.semaphore.sema_thread2thread
- kernel.semaphore.sema_thread2isr
- kernel.semaphore.sema_reset
- kernel.semaphore.sema_count_get

If the testcase is not able to execute for whatever reason, we will have information to report that 4 cases are blocked, i.e. report failure during execution. If one of the tests times out (say the second test case: `kernel.semaphore.sema_thread2isr`) we should be able to report the same, i.e. 1 pass and 3 being blocked.

Sanitycheck will be enhanced to generate the catalog of tests that can be imported into the TMS on a regular basis (moderated) making sure the tests in the source tree and in other locations are in sync with the TMS. How we will moderate the management of testcases and our catalog is TBD, initially however we will manually manage the import of the catalog.

carlescufi, 2018-04-11:
CC @myrdyr

mbolivar, 2018-04-11:
Why are the test cases separated under a `common` key?

Do they have another consumer besides the consumer of the data in the `tests` key?

nashif, 2018-04-11:
> Why are the test cases separated under a common key?

You can have the same tests run under a different condition, for example:

kernel.blah.xxxx

and kernel.blah.poll.xxxx

mbolivar, 2018-04-11:
> You can have the same tests run under a different condition, for example:
> 
> kernel.blah.xxxx
> 
> and kernel.blah.poll.xxxx

Oh, so the cases are globally unique, not just within section.subsection!

That explains why they have a sema_ prefix.

nashif, 2018-04-11:
@mbolivar 
> Oh, so the cases are globally unique, not just within section.subsection!

Take this as an example:
```
common:
  cases:
    - queue_thread2thread
    - queue_thread2isr
    - queue_isr2thread
    - queue_get_2threads
    - queue_get_fail
    - queue_loop
tests:
  kernel.queue:
    tags: kernel
  kernel.queue.poll:
    extra_args: CONF_FILE="prj_poll.conf"
    tags: kernel
```

there will be 2 runs for each test, for example:
- `kernel.queue.queue_loop`
- `kernel.queue.poll.queue_loop`

Thats why the list cases is in common, to avoid repeating them per test instance.

mbolivar, 2018-04-13:
Ah, brilliant -- the `extra_args` example really motivates the separation. Nice work.

nashif, 2018-04-25:
so we have decided to parse the testcases instead of listing them in the yaml file, this works nicely and means we will not go out of sync. see related PR.

nashif, 2019-11-21:
outdated and done already, closing

closed 2019-11-21 by nashif