zephyrproject-rtos/zephyr #7013: cleanup device tree warnings on STM32

galak, 2018-04-10:
We need to cleanup the pin-controller nodes to address the following warnings:

```
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@0 has a unit name, but no reg property
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@1 has a unit name, but no reg property
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@1 has a unit name, but no reg property
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@2 has a unit name, but no reg property
96b_argonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart6@0 has a unit name, but no reg property
96b_argonkey.dts_compiled: Warning (avoid_unnecessary_addr_size): unnecessary #address-cells/#size-cells without "ranges" or child "reg" property in /soc/pin-controller
96b_carbon.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
96b_carbon.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
96b_carbon.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
96b_carbon.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@2 has a unit name, but no reg property
96b_carbon.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
96b_carbon.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@0 has a unit name, but no reg property
96b_carbon.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@1 has a unit name, but no reg property
96b_carbon.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart6@0 has a unit name, but no reg property
96b_carbon.dts_compiled: Warning (avoid_unnecessary_addr_size): unnecessary #address-cells/#size-cells without "ranges" or child "reg" property in /soc/pin-controller
96b_neonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
96b_neonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
96b_neonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
96b_neonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@2 has a unit name, but no reg property
96b_neonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
96b_neonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@0 has a unit name, but no reg property
96b_neonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@1 has a unit name, but no reg property
96b_neonkey.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart6@0 has a unit name, but no reg property
96b_neonkey.dts_compiled: Warning (avoid_unnecessary_addr_size): unnecessary #address-cells/#size-cells without "ranges" or child "reg" property in /soc/pin-controller
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@2 has a unit name, but no reg property
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@1 has a unit name, but no reg property
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@2 has a unit name, but no reg property
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@3 has a unit name, but no reg property
disco_l475_iot1.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@0 has a unit name, but no reg property
disco_l475_iot1.dts_compiled: Warning (avoid_unnecessary_addr_size): unnecessary #address-cells/#size-cells without "ranges" or child "reg" property in /soc/pin-controller
dragino_lsn50.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
dragino_lsn50.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
dragino_lsn50.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
dragino_lsn50.dts_compiled: Warning (avoid_unnecessary_addr_size): unnecessary #address-cells/#size-cells without "ranges" or child "reg" property in /soc/pin-controller
nucleo_f030r8.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
nucleo_f030r8.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
nucleo_f030r8.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
nucleo_f030r8.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
nucleo_f030r8.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@1 has a unit name, but no reg property
nucleo_f030r8.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@2 has a unit name, but no reg property
nucleo_f030r8.dts_compiled: Warning (avoid_unnecessary_addr_size): unnecessary #address-cells/#size-cells without "ranges" or child "reg" property in /soc/pin-controller
nucleo_f070rb.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
nucleo_f070rb.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
nucleo_f070rb.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
nucleo_f070rb.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
nucleo_f070rb.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@1 has a unit name, but no reg property
nucleo_f070rb.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@2 has a unit name, but no reg property
nucleo_f070rb.dts_compiled: Warning (avoid_unnecessary_addr_size): unnecessary #address-cells/#size-cells without "ranges" or child "reg" property in /soc/pin-controller
nucleo_f091rc.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller has a reg or ranges property, but no unit name
nucleo_f091rc.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property
nucleo_f091rc.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@1 has a unit name, but no reg property
nucleo_f091rc.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
nucleo_f091rc.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@1 has a unit name, but no reg property
nucleo_f091rc.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@2 has a unit name, but no reg property
[... cut ...]
stm32_mini_a15.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@0 has a unit name, but no reg property
stm32_mini_a15.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart2@1 has a unit name, but no reg property
stm32_mini_a15.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart3@0 has a unit name, but no reg property
stm32_mini_a15.dts_compiled: Warning (avoid_unnecessary_addr_size): unnecessary #address-cells/#size-cells without "ranges" or child "reg" property in /soc/pin-controller
```

galak, 2018-04-10:
@erwango not sure if this is worth deal with now, or if things will change when we add proper pinctrl/mux support to DT and zephyr

erwango, 2018-04-13:
Warning detected when using dtc v1.4.2.
I porpose to address these for pinctrl introduction

pfalcon, 2018-04-23:
Another DTS warning: https://github.com/zephyrproject-rtos/zephyr/issues/7155

erwango, 2018-04-24:
Being fixed with #7168

erwango, 2018-04-24:
Looking to the warnings, I think one of them is abusive, and I propose to derogate:
    "Node [...] has a unit name, but no reg property"
    ePAPR 1.1 section 2.2.1.1 "Node Name Requirements" specifies that any
    node that has a reg property must include a unit address in its name
    with value matching the first entry in its reg property. Conversely,
    if a node does not have a reg property, the node name must not
    include a unit address.
```
usart1_pins_a: usart1@0 {
	rx_tx {
		rx = <STM32_PIN_PB7 (STM32_PINMUX_ALT_FUNC_7 | STM32_PUPDR_NO_PULL)>;
		tx = <STM32_PIN_PB6 (STM32_PINMUX_ALT_FUNC_7 | STM32_PUSHPULL_NOPULL)>;
	};
};
```			

According to the rule, unit name  `usart1@0` should be used only if node a reg property.
If we want to conform to this rule,  there are 2 options:
- add a reg property to the node, it will looks like:
  ```
  usart1_pins_a: usart1@0 {
        reg = <0>;
	rx_tx {
		rx = <STM32_PIN_PB7 (STM32_PINMUX_ALT_FUNC_7 | STM32_PUPDR_NO_PULL)>;
		tx = <STM32_PIN_PB6 (STM32_PINMUX_ALT_FUNC_7 | STM32_PUSHPULL_NOPULL)>;
	};
  };
  ```
  But reg property is not pertinent to describe a pin or led node. 

- loose phandle associated to the node. This is a pity since phandle are quite handy for many operations like assignments or aliases.
  For instance, instead of: 
 ```
  led0 = &green_led_4;
  pinctrl-0 = <&usart2_pins_d>;
  ```
   we'll have to use:  
  ```
  led0 = "/leds/green_4";
  pinctrl-0 = "/soc/pin-controller@4200000/usart2_pins_d";
  ```

As mentioned in #7168, some of these warnings are also present in STM32 Linux dts files, but associated warning do not show up. I've not been able to confirm, but I assume dtc is ran with -q (quiet) option to silent warnings.

I propose to derogate to this rule and use same option (dtc -q) in in zephyr.

galak, 2018-04-24:
> According to the rule, unit name usart1@0 should be used only if node a reg property.
> If we want to conform to this rule, there are 2 options:

what does linux dt do here, does it use something like @0?

galak, 2018-04-24:
> loose phandle associated to the node. This is a pity since phandle are quite handy for many operations like assignments or aliases.

What's the warning we get for this case?

erwango, 2018-04-24:
> what does linux dt do here, does it use something like @0?

Yes, exact same syntax

> What's the warning we get for this case?

stm32l496g_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property

Side note: 
Using alias syntax such as  `led0 = "/leds/green_4";` requires slight update of extract dts_include script.
Using `pinctrl-0 = "/soc/pin-controller@4200000/usart2_pins_d";` seems to have a deeper impact (devicetree.py as first sight).

galak, 2018-04-24:
> What's the warning we get for this case?
> 
> stm32l496g_disco.dts_compiled: Warning (unit_address_vs_reg): Node /soc/pin-controller/usart1@0 has a unit name, but no reg property

Confused, I was asking about the phandle issue.  What warning do we get related for that.

nashif, 2018-11-05:
@galak is this still an issue?

galak, 2019-01-30:
Closing this now, we have a few dtc warnings, but most aren't related STM32 anymore.  Will open an new issue for the dtc warnings we do have.

closed 2019-01-30 by galak