zephyrproject-rtos/zephyr #7032: Have Sphinx search display txt not ReST as results

mgielda, 2018-04-12:
We had this issue in our own documentation once and managed to fix it by some clever Makefile tricks so I thought I'd take this up.

If you make a search in the Zephyr docs, like:

http://docs.zephyrproject.org/search.html?q=device+tree&check_keywords=yes&area=default

You get results in unrendered ReST, which is kind of unreadable (especially for headers). This is because the search looks in the sources directory which contains ReST files and outputs content directly from them. This can be worked around by producing text files instead.

I think @dbkinder is probably the right person to mention here, I saw you are involvedi n most Sphinx-related stuff.

dbkinder, 2018-04-12:
There have been attempts by the community to address this over the years (as noted in https://github.com/sphinx-doc/sphinx/issues/1618 for example) including hacks to patch the problem (such as https://pypi.python.org/pypi/sphinxprettysearchresults).

We've also had a (long standing) issue associated with the zephyrproject.org website to use Google search instead of the docs-restricted Sphinx search so search results would return hits from all zephyrproject.org sites (www.zephyrproject.org, docs.zephyrproject.org, wiki.zephyrproject.org, lists.zephyrproject.org).  You can simulate this by using the google **site:** option, such as in https://www.google.com/search?&q=site:zephyrproject.org+intel

The one nice thing about the current Sphinx search is it's confined to the current doc set version (e.g., if you're viewing an archived version of the docs, say the [1.10 version](http://docs.zephyrproject.org/1.10.0/index.html), the search results returned are within the 1.10 document set and not from other versions.  We can't control that if we're using Google search, and with Google search, we'd have hits from all visible versions of the docs unless we use a robots.txt to restrict Google scans (something that was [rejected when proposed earlier](https://github.com/zephyrproject-rtos/zephyr/pull/4978)).

That said, I'll see what minimal tweak we can do to at least change the display of the raw ReST format from search results.

mgielda, 2018-04-12:
Thanks for the extensive answer @dbkinder - I know there have been attempts to address this. Of course, searching more broadly is not something we can achieve (easily) with Sphinx, but I was thinking of improving the current readability situation if possible. How I solved this in our case was:

1) generate `txt` files with `make text`
2) change the generated files to have a `rst` extension and put them in the `sources` directory (overwriting the original `rst` files)

There may be an easier way to do this, but this just worked. The search results looked nicer. 
A more minimalistic but not so bad solution is to change the CSS for the `.context` class

```css
    white-space: pre;
    font-family: Courier;
```

nashif, 2018-04-21:
fixed

closed 2018-04-21 by nashif

mgielda, 2018-04-23:
@nashif @dbkinder I daresay the `white-space: pre;` is still a good idea, see below:
[image]
[image]