zephyrproject-rtos/zephyr #7048: STM32F4: Tickless Kernel Timekeeping Problem

mrrosen, 2018-04-13:
**Issue**
`k_cycle_get_32` and `k_uptime_get_32` are not working correctly when `CONFIG_TICKLESS_KERNEL` is enabled on some boards.

**Configuration**
Zephyr Version: `v1.11.0` (on tag of same name)
Board: `stm32f429i_disc1`
Timer Driver: `cortex_m_systick`

**Steps to Reproduce**

> cd samples/subsys/shell/shell_module
> [Edit src/main.c:57 to include the following line: k_enable_sys_clock_always_on();]
> mkdir build && cd build
> cmake -DBOARD=stm32f429i_disc1
> make menucoonfig {Select General Kernel Options -> Power Management and General Kernel Options -> Power Management -> Tickless Kernel
> make && make flash
> minicom -D {serial port for stm32f4 board}
> > select kernel
> > uptime
> > uptime
> > cycles
> > cycles

** Console Output **

> shell> select kernel
> kernel> cycles
> cycles: 3000 hw cycles
> kernel> cycles
> cycles: 338629 hw cycles
> kernel> cycles
> cycles: 675635 hw cycles
> kernel> cycles
> cycles: 1011473 hw cycles
> kernel> uptime
> uptime: 9 ms
> kernel> uptime
> uptime: 10 ms
> kernel> uptime
> uptime: 12 ms
> kernel> cycles
> cycles: 2355130 hw cycles
> kernel> uptime
> uptime: 17 ms
> kernel> uptime
> uptime: 19 ms

** Notes **
It appears theres some issues with time keeping when using tickless kernel mode for this board (and likely all SoCs using `cortex_m_systick`?) were even though the Sys Clock is enabled and never disabled, the functions for reporting uptime and cycle counts are wildly off. I hand typed the above messages and I clearly cannot type "uptime" in 2ms. Without tickless kernel, these values are much more reasonable. Even still, if this were to operate correctly, it should either be made more clear in the documentation or it shouldnt be the case at all that the kernel timing functions like `k_uptime_get` and `k_cycle_get` no longer work when running a tickless kernel. While some modules correctly try to handle the lack of time keeping correctly, others do not (it seems like bluetooth uses these functions alot without calling `k_enable_sys_clock_always_on`). I would suggest forcing correct timing unless another Kconfig variable is set so the default would be to only tick at the maximum interval to keep time but to be able to disable that feature if necessary (ie, a tickless kernel with time keeping and one without), as calling `k_enable_sys_clock_always_on` later in the code losses boot time at very least if not more. Basically a Kconfig variable to set `_sys_clock_always_on` explicitly at boottime instead of being forced to set it later; or to just enable it by default and have applications call `k_disable_sys_clock_always_on` explicitly if they dont need kernel timing.

erwango, 2018-04-26:
Remocing the STM32 tag as this occurs also on frdm_k64f

erwango, 2018-05-04:
@MaureenHelm : Is there any kernel guy that would be interested by this one?
(I'll be off next week, so won't progress much. Besides, I think it might be better covered by an expert)

nashif, 2018-08-26:
@andyross can you please take a look at this one?

andyross, 2018-08-27:
This looks like a fairly pervasive problem with our timer drivers and tickless.  I can reproduce it on frdm-k64f, but also on qemu_cortex_m3 (which uses the same cortex_m_systick driver) and qemu_x86 using both the HPET and LOAPIC drivers.

The only platform that gets this even slightly correct is x86/APIC, because the loapic timer driver returns the TSC count for k_cycle_get_32() (and does it correctly) instead of mucking with the timer hardware.

Technically these are all separate bugs: the shell functions above are indeed just wrappers around the cycle/uptime APIs, which are 100% internal to the timer driver.   Each driver has reimplemented the same bug, I'm guessing.  But nothing looks obviously wrong at first glance.  I'm not much of an expert on timer drivers.

agross-linaro, 2018-08-27:
The driver we have in place is pretty awful.  It needs to be refactored, and I haven't had the time to delve into it.  That's my 2 cents after spending a few hours a couple of months ago looking at this.

andyross, 2018-11-15:
This should be fixed now.  Tickless is the default mode on almost all platforms (not ARC, Nios2 or Pulpino yet) and the test suite exercises these entry points extensively.   A quick test in the shell shows what I think you expect.

andyross, 2019-03-04:
Merged long ago.  Hasn't reappeared.  Can't reproduce.  Please reopen if needed.

closed 2019-03-04 by andyross