zephyrproject-rtos/zephyr #7050: tests: sys_mem_pool: Bus fault occurs on ARM boards (frdmk64f and nrf52840_pca10056)

spoorthik, 2018-04-13:
When tests/kernel/mem_pool/sys_mem_pool is run on frdm_k64f and nrf52840_pca10056 platforms, bus fault occurs. I have tested only on only these 2 platforms of ARM.

Please refer the sample log:

```
 ***** Booting Zephyr OS v1.11.0-675-g9c032eb*****
Running test suite test_sys_mem_pool_api
==============================================
starting test - test_sys_mem_pool_alloc_free
***** BUS FAULT *****
  Executing thread ID (thread): 0x20007c84
  Faulting instruction address:  0x6663742c
  Instruction bus error
Fatal fault in thread 0x20007c84! Aborting.
```

Whereas the test passes on X86, Xtensa, nios2, riscv32 boards.
Sample log below:
```
 Running test suite test_sys_mem_pool_api
 ===================================================================
 starting test - test_sys_mem_pool_alloc_free
 PASS - test_sys_mem_pool_alloc_free.
 ===================================================================
 starting test - test_sys_mem_pool_alloc_align4
 PASS - test_sys_mem_pool_alloc_align4.
 ===================================================================
 starting test - test_sys_mem_pool_min_block_size
 PASS - test_sys_mem_pool_min_block_size.
 ===================================================================
 ===================================================================

 PROJECT EXECUTION SUCCESSFUL
```

MaureenHelm, 2018-05-01:
The test doesn't fail on current master (7be3236) on frdm_k64f. @spoorthik, do you know which commit this failed on?

spoorthik, 2018-05-09:
@MaureenHelm : I dont see this issue with the latest commit as you said. Let this issue be closed. I will reopen with the right commit ID.

closed 2018-05-09 by spoorthik