zephyrproject-rtos/zephyr #7054: Zephyr's Kconfig implementation has an undocumented divergence

SebastianBoe, 2018-04-13:
Zephyr's Kconfig implementation diverges from Linux by prefering later defaults. This is a problem for several reasons:

### Problem description

Zephyr can no longer rely on the existing Kconfig ecosystem for documentation. Everything in the Linux kernel's Kconfig documentation applies to Zephyr's Kconfig except that the documentation is wrong when it states that earlier defaults have precedence.

The divergence must be implemented in Kconfig tools. Currently this is done with an out-of-tree patch on kconfiglib that is expensive to maintain.

Readers already familiar with the original Kconfig language will read Zephyr Kconfig files incorrectly. The above two issues are resolvable without aligning with Linux, but this is not.

### Problem solutions

The cleanest solution is simply to follow the Linux kernel's behaviour. But to do this
it must be better understood why the divergence occured and what the alternatives for diverging were. @ulfalizer has already looked into this here (https://github.com/zephyrproject-rtos/zephyr/issues/6890#issuecomment-377946637), so we have some idea of why the divergence occured, but our options for re-aligning are not well understood. 

Alternatively, one could keep the divergence, and maintain a fork of the Kconfig language documentation in Zephyr. A fork of the documentation is presumably going to be necessary anyway since we intend to start taking advantage of Kconfiglib extentions that are not used or documented by Linux.

DoD: There exists documentation that matches Zephyr's Kconfig implementation.

ulfalizer, 2018-04-26:
The documentation added in https://github.com/zephyrproject-rtos/zephyr/pull/7199 mentions that Zephyr's Kconfig prefers later defaults. Do you think that's enough?

Might be easy to miss, but it's a big improvement from nothing at least. :)

SebastianBoe, 2018-04-27:
It helps :)

But it would still be better to avoid this patch. I will re-write this issue from a doc bug to an enhancement.

SebastianBoe, 2018-04-27:
@ulfalizer : Any ideas on how to align?

I assume that simply re-ordering and doing:

> config FOO
>   default 42
> 
> config FOO
>   int
>   default 32
> 

instead of

> config FOO
>   int
>   default 32
> 
> config FOO
>   default 42
> 

is not valid?

ulfalizer, 2018-04-27:
@SebastianBoe 
Yeah, I've also been thinking about whether you can simply switch them around. Think it might work.

One messy thing is that many symbols already have many `default`s in the same location and depend on the prefer-later-defaults behavior. Might be a lot of stuff that would need to be fixed.

SebastianBoe, 2018-10-09:
This has been fixed.

closed 2018-10-09 by SebastianBoe