zephyrproject-rtos/zephyr #7055: tests: fatal: Stack protection fatal test fails on ARC core

spoorthik, 2018-04-13:
**Test location:** zephyr.git/tests/kernel/fatal-stack.protection

**Board used:** Arduino_101_sss, quark_se_c1000_ss_devboard

Please refer the sample log below:

```
***** Booting Zephyr OS 1.11.99  *****
Running test suite fatal
===================================================================
starting test - test_fatal
test alt thread 1: generic CPU exception
Exception vector: 0x8, cause code: 0x0, parameter 0x0
Address 0x400347cc
Current thread ID = 0xa8000400
Faulting instruction address = 0x400347cc
Caught system error -- reason 0
test alt thread 2: initiate kernel oops
@ zephyr.git/tests/kernel/fatal/src/main.c:95:
***** Kernel OOPS! *****
Current thread ID = 0xa8000400
Faulting instruction address = 0x40034c68
Caught system error -- reason 4
test alt thread 3: initiate kernel panic
@ zephyr.git/tests/kernel/fatal/src/main.c:106:
***** Kernel Panic! *****
Current thread ID = 0xa8000400
Faulting instruction address = 0x40034c68
Caught system error -- reason 5
test stack overflow - timer irq
posting 1024 bytes of junk to stack...
busy waiting...
FAIL - stack_thread1@126. should never see this
@ zephyr.git/kernel/thread.c:183:
**** Unknown Fatal Error 2! ****
Current thread ID = 0xa8000400
Faulting instruction address = 0x400359e8
Caught system error -- reason 2

    Assertion failed at zephyr.git/tests/kernel/fatal/src/main.c:225: test_fatal: (rv equal to TC_FAIL)
thread was not aborted
FAIL - test_fatal.
===================================================================
===================================================================
```

**Steps to reproduce:**
```
cd zephyr.git/tests/kernel/fatal
mkdir build && cd build 
cmake -DBOARD=arduino_101_sss .. 
make run
```

vonhust, 2018-04-16:
This bug can be fixed by PR #7036

nashif, 2018-04-21:
#7036 merged already

closed 2018-04-21 by nashif