zephyrproject-rtos/zephyr #7067: scripts: extract_dts_includes: fails on multiple includes in yaml bindings file

b0661, 2018-04-13:
Inheriting from multiple bindings fails.

    inherits:
        !include [binding1.yaml, binding2.yaml]

Following the above the 'node_type' node property does not account for multiple base types.

Loader.include for yaml.SequenceNode and yaml_traverse_inherited seems to be broken.

galak, 2018-04-18:
@erwango can you look into this issue.

erwango, 2018-04-18:
@b0661 : In order to solve this is a coherent way, do you have any example of use for multiple bindings inclusion?

erwango, 2018-05-03:
@b0661 : Is that ok if we set this ticket as 'enhancement' instead of 'bug'? (Medium priority 'bug' issues are blocking for upcoming release)
This is blocking further advanced development but not breaking anything currently.

b0661, 2018-05-03:
@erwango 

>Is that ok if we set this ticket as 'enhancement' instead of 'bug'?

Yes, shure it is ok to set it to 'enhancement'.

PR #7548 (2018-05-25): scripts: extract_dts_includes.py: fix multiple include in bindings
Correctly process multiple include files given to the
!include command of the YAML loader.

The fix only targets the sequential definition of include files.

The PR changes the 'node_type' node property to list type to account for multiple base types.

Fixes #7067
commit 58967c7d (2018-05-25): scripts: extract_dts_includes.py: fix multiple include in bindings

Correctly process multiple include files given to the
!include command of the YAML loader.

The fix only targets the sequential definition of include files.

Fixes #7067
closed 2018-05-25 by galak