zephyrproject-rtos/zephyr #7075: No such file or directory: CMAKE_READELF: 'CMAKE_READELF-NOTFOUND'

gyboy, 2018-04-15:
when I tried to config the board ID to run example,something wrong with cmake,it seems that the gcc based toolchains didn't work well. I was just cloned the master to my ubuntu and build cmake-3.8.2 then installed,what's the problem?the log show below:
CMake Error at /home/ubuntu/Desktop/zephyr/cmake/extensions.cmake:984 (message):
  No such file or directory: CMAKE_READELF: 'CMAKE_READELF-NOTFOUND'
Call Stack (most recent call first):
  /home/ubuntu/Desktop/zephyr/cmake/compiler/gcc.cmake:14 (assert_exists)
  /home/ubuntu/Desktop/zephyr/cmake/toolchain.cmake:38 (include)
  /home/ubuntu/Desktop/zephyr/cmake/app/boilerplate.cmake:243 (include)
  CMakeLists.txt:1 (include)

lpereira, 2018-04-16:
Seems to be a duplicate of #6911.

andrewboie, 2018-04-16:
This happens when the build system cannot find the toolchain.
Check your environment and build variable settings.
Are you building on OS X?

nashif, 2018-04-18:
This is one of those misleading warnings or errors, as @andrewboie said, it is not finding the toolchain, so please make sure you have all env variables set correctly and make sure you have a toolchain available.

SebastianBoe, 2018-04-19:
"when I tried to config the board ID to run example,something wrong with cmake,it seems that the gcc based toolchains didn't work well. I was just cloned the master to my ubuntu and build cmake-3.8.2 then installed,what's the problem?"

Please follow the getting started guide and post the environment variables that are set.

SebastianBoe, 2018-04-19:
Please run

echo $ZEPHYR_SDK_INSTALL_DIR
echo $ZEPHYR_TOOLCHAIN_VARIANT

closed 2018-04-19 by nashif

gyboy, 2018-04-21:
I modified the gcc.make file and run "cmake -GNinja -DBOARD=arduino_101 ../.." again ,it remind me that
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD arduino_101
Zephyr version: 1.11.99
Parsing Kconfig tree in /home/ubuntu/Desktop/zephyr/Kconfig
Using /home/ubuntu/Desktop/zephyr/boards/x86/arduino_101/arduino_101_defconfig as base
Merging /home/ubuntu/Desktop/zephyr/samples/hello_world/prj.conf
CMake Error at /home/ubuntu/Desktop/zephyr/cmake/compiler/gcc.cmake:17 (message):
  Zephyr was unable to find the toolchain.  Is the environment misconfigured?

  User-configuration:

  ZEPHYR_TOOLCHAIN_VARIANT: zephyr

  Internal variables:

  CROSS_COMPILE:
  /home/ubuntu/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elfiamcu/i586-zephyr-elfiamcu-

  TOOLCHAIN_HOME: /home/ubuntu/sysroots/x86_64-pokysdk-linux

Call Stack (most recent call first):
  /home/ubuntu/Desktop/zephyr/cmake/toolchain.cmake:38 (include)
  /home/ubuntu/Desktop/zephyr/cmake/app/boilerplate.cmake:243 (include)
  CMakeLists.txt:2 (include)

-- Configuring incomplete, errors occurred!

it seem that the ZEPHYR_TOOLCHAIN_VARIANT  was correctly what I have set, I am sorry that I am not familiar with linux develop environment, what should I do to correct it?

SebastianBoe, 2018-04-23:
@gyboy : I'm not sure what the root cause is.

One abnormality is that sysroots is in  /home/ubuntu/, it should be in an "sdk" directory within /home/ubuntu/ instead.

This is what it looks like for me:

> sebo@mach:~/zephyr$ ls ~/zephyr-sdk-0.9.2-rc5/
> info-zephyr-sdk-0.9.2  sdk_version  sysroots
> 

As you can see, the SDK should have a file next to sysroots called sdk_version, this file is now in your home directory, which is not great, although I can't see why this would actually be causing any errors.

Could you try re-installing the SDK? This time to a dedicated folder within /home/ubuntu.

Also, could you check if the file /home/ubuntu/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elfiamcu/i586-zephyr-elfiamcu-readelf

and /home/ubuntu/sysroots/x86_64-pokysdk-linux/usr/bin/i586-zephyr-elfiamcu/i586-zephyr-elfiamcu-gcc

exist and find out if they are executable.

gyboy, 2018-04-25:
@SebastianBoe,thanks,actually I didn't find the directory.It seems that I have installed the sdk for many times,the first time I used the default directory whihout the command sudo and I can't see any file in  "/opt/zephyr sdk",the second time I installed the sdk in the directory "/home/ubuntu/",which you can see from the log,but still can't find any file,the last time I installed in "/home/ubuntu/zephyr-sdk",still didn't work.every time  I change the respond path of "ZEPHYR_SDK_INSTALL_DIR". I was considering whether should I remove the last time installed file then excute the setup again,however I don't know how to remove the installed sdk. I'll try to reconfig the whole environment as the guide have instructed, thanks a lot

SebastianBoe, 2018-04-26:
Resolved.

PR #7117 (2018-04-19): cmake: toolchain: Improve error feedback when toolchain is not found
Improve the error feedback when the toolchain is not found.

Currently, for some setups, presumably primarily SDK-less setups,
users can get an error message like this for certain types of
misconfigurations:

CMake Error at /home/ubuntu/Desktop/zephyr/cmake/extensions.cmake:984 (message):
  No such file or directory: CMAKE_READELF: 'CMAKE_READELF-NOTFOUND'
Call Stack (most recent call first):
  /home/ubuntu/Desktop/zephyr/cmake/compiler/gcc.cmake:14 (assert_exists)
  /home/ubuntu/Desktop/zephyr/cmake/toolchain.cmake:38 (inclu [... cut]
commit f10fddeb (2018-04-19): cmake: toolchain: Improve error feedback when toolchain is not found

Improve the error feedback when the toolchain is not found.

Currently, for some setups, presumably primarily SDK-less setups,
users can get an error message like this for certain types of
misconfigurations:

CMake Error at zephyr/cmake/extensions.cmake:984 (message):
  No such file or directory: CMAKE_READELF: 'CMAKE_READELF-NOTFOUND'
Call Stack (most recent call first):
  zephyr/cmake/compiler/gcc.cmake:14 (assert_exists)
  zephyr/cmake/toolchain.cmake:38 (include)
  zephyr/cmake/app/boilerplate.cmake:243 (include)
  CMake [... cut]