zephyrproject-rtos/zephyr #7076: NRF52840: I2C Sensor (SHT3XD) driver issue

IotPanorama, 2018-04-15:
I am trying to use I2C SHT3XD sensor with nrf52840. i am getting the following fatal fault.
***** BOOTING ZEPHYR OS v1.11.0 - BUILD: Apr 15 2018 10:42:29 *****
dev 0x20002be8 name SHT3XD
***** BUS FAULT *****
  Executing thread ID (thread): 0x20000ce4
  Faulting instruction address:  0x4f480588
  Instruction bus error
Fatal fault in essential thread! Spinning...

IotPanorama, 2018-04-16:
Please, someone take a look at that. I am sort of stuck here.

MaureenHelm, 2018-04-17:
What board are you using? Have you written your own application or are you using an in-tree sample application?

IotPanorama, 2018-04-17:
I am using nrf52840 pdk. and actually i am using in-tree sample application bme280 with change in prj.conf.

anangl, 2018-05-07:
Perhaps this sensor tries to use a bus that is not configured in your app.

I used the "bme280" sample on the pca10056 board and got the same error.
After adding the following to `prj.conf`:
```
CONFIG_SYS_LOG=y
CONFIG_SYS_LOG_SENSOR_LEVEL=4
```
I could see this additional message:
```
[BME280] [DBG] bme280_init: i2c master not found: I2C_0
```

So I added to `prj.conf`:
```
CONFIG_I2C_0=y
CONFIG_I2C_NRF5=y
CONFIG_I2C_NRF5_0_GPIO_SDA_PIN=26
CONFIG_I2C_NRF5_0_GPIO_SCL_PIN=27
```
and then the app started the I2C communication successfully (I haven't got the sensor, so I get only zeros, but the fault is gone).

Does this work for you as well?

IotPanorama, 2018-05-14:
After adding the configuration parameters i got a build error (screen shot has been attached
[image]
) .
So I added to following in prj.conf
CONFIG_I2C_0_IRQ_PRI=7

Now Its working completely fine.

anangl, 2018-05-15:
@IotPanorama Can we close this issue then?

closed 2018-05-15 by IotPanorama