zephyrproject-rtos/zephyr #7091: [Coverity CID:185273] Resource leaks in /tests/net/socket/getaddrinfo/src/main.c

Shilpashree3lc, 2018-04-16:
Static code scan issues seen in File: /tests/net/socket/getaddrinfo/src/main.c
 Category: Resource leaks
 Function: test_getaddrinfo_ok
 Component: Tests
 CID: 185273
 Please fix or provide comments to square it off in coverity in the link: https://scan9.coverity.com/reports.htm#v32951/p12996

pfalcon, 2018-04-19:
@stig-bjorlykke: As the author of this test, would you like to have a look/try to do something? If there're any issues, just let me know and I'll cover it. Thanks.

closed 2018-04-20 by jukkar

pfalcon, 2018-04-20:
> I suspect this is because coverity expects the addrinfo returned from getaddrinfo() to be freed with freeaddrinfo(). In Zephyr this is not strictly needed, but I will provide a fix for this test to be coverity compliant.

That would also be my solution, glad we're on the same line here, and thanks for sharing the maintenance load. Please keep in mind that there's always a choice to mark a Coverity warning as "false positive", and in tests, being formally correct is indeed not that important. But if a better fix is simple and doesn't affect the flow of test, indeed, why not fix it, instead of silencing on Coverity side?

PR #7133 (2018-04-20): tests: socket: Free resources with freeaddrinfo
In Zephyr this has no effect because getaddrinfo() returns a pointer
to a static array, but Coverity scan checks for this pattern.

Coverity-CID: 185273
Coverity-CID: 185279

Fixes #7085
Fixes #7091
commit e8bcc6f1 (2018-04-20): tests: socket: Free resources with freeaddrinfo

In Zephyr this has no effect because getaddrinfo() returns a pointer
to a static array, but Coverity scan checks for this pattern.

Coverity-CID: 185273
Coverity-CID: 185279

Fixes #7085
Fixes #7091