zephyrproject-rtos/zephyr #7120: Kconfig sources of SoC's should not be able to affect another

SebastianBoe, 2018-04-19:
HW platforms are affecting each other's Kconfig more than they need be. This has an adverse usability side-effect.

The current Kconfig implementation behaves like this:

The user specifies a BOARD. Since boards are organized into zephyr/boards/$ARCH/<some_board> the build system is able to infer which architecture the board belongs to, but not for instance which SoC it belongs to. It then gives input[0] to Kconfig that we are using such-and-such board and such-and-such ARCH.

Kconfig then uses this HW information to prune away[1] Kconfig sources that are irrelevant to the board and arch. This pruning was initially implemented for performance reasons, but this isolation also has other benefits. For instance, poorly written Kconfig sources in one architecture will not cause warnings in another architecture[2].

Definition Of Done: Extend the isolation of Kconfig sources to also cover SoC's. Alternatively, to the extent that is practically possible prune all Kconfig sources that are not relevant to the BOARD.

[0] https://github.com/zephyrproject-rtos/zephyr/blob/28724732e7dc9b9aa595ac4f4c772f35f310724e/cmake/kconfig.cmake#L48

[1] https://github.com/zephyrproject-rtos/zephyr/blob/28724732e7dc9b9aa595ac4f4c772f35f310724e/Kconfig.zephyr#L46

[2] https://github.com/zephyrproject-rtos/zephyr/issues/7097

SebastianBoe, 2018-04-19:
@ulfalizer : Not sure if there is anything to do on the Kconfig side. Perhaps this is a pure integration issue. But any input would be appreciated.

SebastianBoe, 2018-04-19:
Thinking out loud.

I wonder if things would work better if the build system did this instead:

Use the BOARD input to select an appropriate DTS.
Use the DTS output to detect ARCH, SOC, and other HW information.

Feed the HW information from DTS into Kconfig to provide the user with reasonable defaults and prevent the user from configuring something that is not configurable. E.g. the user should not be able to configure that ADC should be used if the SoC does not have an ADC and vice versa.

ulfalizer, 2018-04-19:
@SebastianBoe 
Pruning on the Kconfiglib side gets tricky, so it would probably be best to do it on the `cmake` side.

You can't know in general whether an `if FOO  ...  endif` will be true or false while parsing for example, since the `FOO` symbol might be defined after the `if` (or depend on things defined after the `if`). The only really sane thing you can do in Kconfig as it is is to parse everything.

SebastianBoe, 2018-04-19:
Would it be possible to have a symbol type that was immutable and required forward-declaration? The typical use-case is symbols describing HW.

Then Kconfig could first source HW symbols and then source SW Kconfig sources.

ulfalizer, 2018-04-19:
@SebastianBoe 
One thing that would make environment variables cleaner (and that's also planned for the Linux Kconfig) would be to get rid of those `option env` symbols, and just have `$FOO` automatically reference environment variables.

Those "bounce symbols" always felt kinda pointless to me, so it's a good change, imo.

ulfalizer, 2018-04-19:
On Linux, there's just a separate top-level `Kconfig` file per architecture. That one in turn pulls in a bunch of `Kconfig` files, many of which are shared between architectures.

SebastianBoe, 2018-04-19:
> One thing that would make environment variables cleaner (and that's also planned for the Linux Kconfig) would be to get rid of those option env symbols, and just have $FOO automatically reference environment variables.

That would work. No need for immutable symbols then.

"On Linux, there's just a separate top-level Kconfig file per architecture. That one in turn pulls in a bunch of Kconfig files, many of which are shared between architectures."

Okay, so they use the same solution and are affected by the same problem as us:
https://github.com/torvalds/linux/blob/master/arch/arm/Kconfig#L713

Good to know.

carlescufi, 2018-05-07:
@ulfalizer is this something you can look into after the menuconfig work + getting rid of C tools + using `rsource` and `gsource` is complete?

carlescufi, 2018-09-14:
@ulfalizer can you take another look at this one when you have a chance?

ulfalizer, 2019-03-07:
> One thing that would make environment variables cleaner (and that's also planned for the Linux Kconfig) would be to get rid of those `option env` symbols, and just have `$FOO` automatically reference environment variables.

This has been done now at least.

Don't know if there are any great issues with SoCs affecting each other anymore. Might require some pretty deep reorganization to do something about. Closing for now. Feel free to reopen.

closed 2019-03-07 by ulfalizer

SebastianBoe, 2019-03-07:
This enhcancement is still valid and should be re-opened.