zephyrproject-rtos/zephyr #7137: I2C Driver does not compile for nrf5 boards

hakonfam, 2018-04-20:
When enabling I2C driver for nrf52_pca10040 or nrf51_pca10028 the following error is given:

```
zephyr/include/generated/generated_dts_board.h:91:33: error: 'NORDIC_NRF5_I2C0_40003000_BASE_ADDRESS_0' undeclared here (not in a function); did you mean 'NORDIC_NRF_UART_40002000_BASE_ADDRESS_0'?
zephyr/include/generated/generated_dts_board.h:93:31: error: 'NORDIC_NRF5_I2C0_40003000_CLOCK_FREQUENCY' undeclared here (not in a function); did you mean 'NORDIC_NRF5_I2C0_40003000_BASE_ADDRESS_0'?
zephyr/include/generated/generated_dts_board.h:92:28: error: 'NORDIC_NRF5_I2C0_40003000_LABEL' undeclared here (not in a function); did you mean 'NORDIC_NRF_UART_40002000_LABEL'?
```
And so on.

Tested on version c0d0a61bcc4f
To reproduce, build `samples/bluetooth/beacon/` for either nrf52_pca10040 or nrf51_pca10028 with the following configuration set:

CONFIG_I2C=y
CONFIG_I2C_NRF5=y
CONFIG_I2C_0=y
CONFIG_I2C_NRF5_0_GPIO_SDA_PIN=12
CONFIG_I2C_NRF5_0_GPIO_SCL_PIN=13

hakonfam, 2018-04-20:
It looks like putting 
```
&i2c0{
    status = "ok";
    compatible = "nordic,nrf5-i2c";
}; 
```

into `boards/arm/nrf52_pca10040/nrf52_pca10040.dts` fixes the issue, although I do not know why yet.

SebastianBoe, 2018-04-20:
@anangl : AFAICT from the description this means that I2C is broken on nRF devices for release 1.11.

hakonfam, 2018-04-20:
It looks like 51ba17 and 422bc2 moved the I2C configuration into DTS, but forgot to enable I2C on the board specific dts files.

PR #7139 (2018-04-23): boards: dts: Add i2c to nrf5X_pcaX board dts.
Fixes #7137

Enables the i2c drivers for these boards, allowing I2C to be used.
Required to enable i2c with these boards.
Tested building samples/bluetooth/beacon for nrf52_pca10040 with i2c enabled.

Parameters used for enabling i2c:
CONFIG_I2C=y
CONFIG_I2C_0=y
CONFIG_I2C_NRF5=y
CONFIG_I2C_NRF5_0_GPIO_SCL_PIN=6
CONFIG_I2C_NRF5_0_GPIO_SDA_PIN=7
commit 597517c7 (2018-04-23): boards: dts: Add i2c to nrf5X_pcaX board dts.

Fixes #7137

Enables the i2c drivers for these boards.
Required to enable i2c with these boards.
Tested samples/bluetooth/beacon for nrf52_pca10040
with i2c enabled.

Parameters used for enabling i2c:
CONFIG_I2C=y
CONFIG_I2C_0=y
CONFIG_I2C_NRF5=y
CONFIG_I2C_NRF5_0_GPIO_SCL_PIN=6
CONFIG_I2C_NRF5_0_GPIO_SDA_PIN=7
closed 2018-04-23 by galak