zephyrproject-rtos/zephyr #7151: boards: Move existing boards to "default configuration guidelines"

erwango, 2018-04-23:
Following introduction of "default board configuration guidelines" (#6858), one part of the work is to move boards already ported to zephyr to conform with guidelines available here:
http://docs.zephyrproject.org/porting/board_porting.html#default-board-configuration

In order to migrate all boards to the new standards, work is split into several tickets. See below.
As part of the work to do on each board:
- pins should be configured to fit valuable HW components available on board
- pins should be configured to fit connector if featured on the board.
- propose a default network interface (when possible)
- If ever a pin conflict prevents configuration of one of the components, board documentation should be updated with conflict and arbitration.
- update board metadata yaml file vs new board configuration [(cf documentation)](https://github.com/zephyrproject-rtos/zephyr/blob/master/doc/subsystems/test/sanitycheck.rst#board-configuration)

In order to allow provision of dts connector aliases, dtc > v1.4.1 should be used. Hence all these tickets (at least when a connector is available on board) have a dependency on zephyr SDK v0.9.3:
https://github.com/zephyrproject-rtos/meta-zephyr-sdk/releases/tag/0.9.3-rc2

- [ ] Get 96Boards boards (4) compliant with default configuration guidelines
- [ ] Get ARC boards (4) compliant with default configuration guidelines
- [ ] Get XTENSA boards (2) compliant with default configuration guidelines
- [ ] Get X86 boards (8) compliant with default configuration guidelines
- [ ] Get RISCV boards (5) compliant with default configuration guidelines
- [ ] Get POSIX board compliant with default configuration guidelines
- [ ] Get NIOS boards (2) compliant with default configuration guidelines
- [ ] Get ARM boards (2) compliant with default configuration guidelines
- [ ] Get ATMEL SoC based boards (6) compliant with default configuration guidelines
- [ ] Get NORDIC SoC based boards (15) compliant with default configuration guidelines
- [ ] Get NXP SoC based boards (8) compliant with default configuration guidelines
- [ ] Get SILABS board compliant with default configuration guidelines
- [ ] Get TI SoC based boards (4) compliant with default configuration guidelines
- [ ] Get ST Nucleo boards (14) compliant with default configuration guidelines
- [ ] Get ST Disco and Eval boards (14) compliant with default configuration guidelines
- [ ] Get Olimex boards (4) compliant with default configuration guidelines
- [ ] Get ST Based boards (3) compliant with default configuration guidelines

erwango, 2018-04-23:
@nashif , @galak : follow up of [Default board configuration guidelines](#6858).
I'll start deploying if you're ok with this.

erwango, 2018-04-23:
> I'll start deploying if you're ok with this.

I'll provide a reference PR beforehand

pizi-nordic, 2018-08-01:
FYI: @carlescufi, @anangl

closed 2021-04-26 by erwango