zephyrproject-rtos/zephyr #7159: Kconfig.defconfig is undocumented and unclear

ulfalizer, 2018-04-23:
`Kconfig.defconfig` files were mysterious to me, meaning they're probably super mysterious to people who are less used to Kconfig. They ought to be documented, both to make the design clearer, and to prevent people from putting stuff in random places.

The prefer-later-defaults behavior (see https://github.com/zephyrproject-rtos/zephyr/issues/7054) could be documented at the same time.

Here's an overview of what I think the thinking behind those files is, taken from https://github.com/zephyrproject-rtos/zephyr/issues/6890#issuecomment-377946637:

 1. Boards have some fixed setting/symbol X.

  2. If a user picks a particular board, they shouldn't be able to change X in e.g. `menuconfig`, so X has no `prompt`.

  3. Since X has no prompt, it can't be assigned its board-specific value via a `.config` (`*_defconfig` in this case) file.

  4. X still needs to get its board-specific value somehow though, and that's what those `Kconfig.defconfig` files are for. You could have a single symbol definition with all the `default`s too, but it would get spammy.

It would be a good idea to have that documented I think. Otherwise, people are likely to put stuff in random places.

If I've understood the design right, then I guess the "proper" fix would be to give the symbol a prompt in one of those arch- or board-specific `Kconfig` files, if it should be user-configurable on that particular machine.

As a note, you can put a condition on a prompt too, like `prompt "foo" if X`. If `X` is `n`, the symbol won't be user-configurable ("visible", in Kconfig speak).

ulfalizer, 2018-04-23:
CC @carlescufi

SebastianBoe, 2018-04-24:
I don't know what the design behind Kconfig.defconfig is either.

But one would think that the use-case for having board-specific immutable/fixed configs should be covered by DTS and then input to Kconfig like ARCH is.

carlescufi, 2018-04-24:
Agreed, these files are a mystery. I've assigned you to this @ulfalizer so you can document them.

PR #7199 (2018-04-27): doc: Document Zephyr's Kconfig configuration scheme
In particular, try to demystify `Kconfig.defconfig` files, and provide
guidelines on whether configuration settings should go in
`BOARD_defconfig` or `Kconfig.defconfig`.

The board porting section seems to be the most relevant one here, so put
the documentation there, with some "see also" links from elsewhere.
Things could be reorganized later if needed.

Give a general overview of visible and invisible Kconfig symbols as
well, as that ties in with the configuration scheme.

This is reverse-engineering on my part. The configuration scheme do [... cut]
commit 63cb2334 (2018-04-27): doc: Document Zephyr's Kconfig configuration scheme

In particular, try to demystify Kconfig.defconfig files, and provide
guidelines on whether configuration settings should go in
BOARD_defconfig or Kconfig.defconfig.

The board porting section seems to be the most relevant one here, so put
the documentation there, with some "see also" links from elsewhere.
Things could be reorganized later if needed.

Give a general overview of visible and invisible Kconfig symbols as
well, as that ties in with the configuration scheme.

This is reverse-engineering on my part. The configuration scheme doesn't [... cut]
closed 2018-04-27 by carlescufi