zephyrproject-rtos/zephyr #7172: Etherent initialization crashes in frdm_k64f

rveerama1, 2018-04-24:
Ethernet initialization crashes in frdm_k64f, last upstream commit I noticed this crash is
--------------------------------------------------------------------------------------
commit f5f95ee3a909e522acb1a8ee534635d45803441c
Author: Leandro Pereira <leandro.pereira@intel.com>
Date:   Fri Apr 6 15:55:11 2018 -0700

    kernel: sem: Ensure that initial count is lesser or equal than limit
--------------------------------------------------------------------------------------

[dev/eth_mcux] [DBG] eth_0_init: MAC 00:04:9f:1e:c8:3e
***** BUS FAULT *****
  Executing thread ID (thread): 0x20008a14
  Faulting instruction address:  0x11f38020
  Instruction bus error
Fatal fault in essential thread! Spinning...

rveerama1, 2018-04-24:
@mjs-arm @MaureenHelm assign to right person if it is not related to you.

MaureenHelm, 2018-04-24:
Thanks @mike-scott for reporting the additional failures. I was thinking about fixing the mcr20a driver by adding dts bindings, but with the additional dependencies in the sensor drivers, I'll revert the bad PR.

PR #7355 (2018-05-10): dts: Add sensor and mcr20a bindings, and fix samples
- Adds dts bindings for the fxos8700 and fxas21002 interrupt pins
- Refactors mcux dspi driver to dts
- Adds dts bindings for mcr20a

Fixes #7172 

Pulls in changes from #7350, #7189, and #7105
closed 2018-05-10 by galak