zephyrproject-rtos/zephyr #7179: _vprintk incorrectly prints 64-bit values

d3zd3z, 2018-04-24:
`_vprintk` and by extension all of the formatted print routines in Zephyr support `"%lld"` and variants to print 64-bit values. However, the code that does this discards the top 32-bits of these values, and just prints them using the 32-bit printing code:
``` c
                                long d;
                                if (long_ctr < 2) {
                                        d = va_arg(ap, long);
                                } else {
                                        d = (long)va_arg(ap, long long);
                                }
```

This could be fixed by by generalizing the print routines to use 64-bit values, although there could be a performance cost to this affecting all of the print routines. However, the current behavior is misleading, and we'd probably be better printing something like "ERR" rather than printing incorrect values.

d3zd3z, 2018-04-24:
@andrewboie this does still print garbage for 64-bit values that are larger than 32-bit.

d3zd3z, 2018-05-22:
Also, note that `tests/kernel/common/src/printk.c` tests for the incorrect behavior, asserting that printing '-1' as an unsigned is `4294967295`, and `ffffffff` in hex. There are some other places in the code that make assumptions about the length of the maximum value, and buffers overflow.

andrewboie, 2019-01-17:
I fixed printing 64-bit hex values. I left base-10 values alone but I can look at it.

PR #12552 (2019-01-18): printk: don't print incorrect 64-bit integers
printk is supposed to be very lean, but should at least not
print garbage values. Now when a 64-bit integral value is
passed in to be printed, 'ERR' will be reported if it doesn't
fit in 32-bits instead of truncating it.

The printk documentation was slightly out of date, this has been
updated.

Fixes: #7179
commit 97075840 (2019-01-18): printk: don't print incorrect 64-bit integers

printk is supposed to be very lean, but should at least not
print garbage values. Now when a 64-bit integral value is
passed in to be printed, 'ERR' will be reported if it doesn't
fit in 32-bits instead of truncating it.

The printk documentation was slightly out of date, this has been
updated.

Fixes: #7179
closed 2019-01-18 by andrewboie