zephyrproject-rtos/zephyr #7193: tickless and timeslicing don't play well together

andyross, 2018-04-25:
Specific bug: wake up a thread (or create a new one) at the same priority as the currently running thread, both of which are at the same preemptible priority and in a situation where no other threads exist at that priority.  In theory, the CPU timer should be adjusted to reflect the new situation and expire in one slicing tick.  In practice, the timer is not adjusted and the new thread won't get any CPU until whenever the next interrupt fires, which could be a long time.

That issue was found via inspection as I stumpled over this code while working on the scheduler.  The more general problem is that we have pretty poor test coverage for both timeslicing and tickless behaviors, this situation needs to be tested and handled, along with (probably) others.

jli157, 2018-04-26:
Could the issue https://github.com/zephyrproject-rtos/zephyr/issues/7048 be caused by same root cause?

andyross, 2018-07-24:
The bug found by inspection will be fixed.  Probably need some sort of longer term tracking on enhancing tickless (making it default where possible, expanding tests) and coming up with a better set of tests on timeslicing that look for fairness issues like this.

pizi-nordic, 2018-07-26:
FYI: @pizi-nordic

andyross, 2018-08-23:
Fix for the specific bug above is submitted.  Probably should reprioritize rather than close this one, as the rework is if anything needed more now than before, given that we've added another #ifdef TICKLESS case.

andyross, 2018-08-27:
Ugh, fixed correctly this time.   No idea what happened earlier, presumably I rebased, but then forgot the "-a" on the commit?

andyross, 2018-11-15:
Fixed as of the big timer rework.  Tickless is default and all timeslicing tests pass with it enabled. Closing as I'm the one who opened it to begin with.

closed 2018-11-15 by andyross