zephyrproject-rtos/zephyr #7198: sanitycheck issue w/ztest

galak, 2018-04-26:
The following commit added some checks which case break on tests/kernel/mem_protect/mem_protect/src/main.c, because of the #ifndef/#endif block.

commit aae71d74dd2f26f970f8dc0479ea706fde36dbdc
Author: Anas Nashif <anas.nashif@intel.com>
Date:   Sat Apr 21 22:26:48 2018 -0500

    sanitycheck: parse test cases from source files
    
    This parses the tests that run within a test project/application from
    the source code and gives us a view of what was run, skipped and what
    was blocked due to early termination of the test.

galak, 2018-04-26:
I think this shows two different issues.

1. The error reporting of sanitycheck isn't for parsing the test case isn't doing the right thing
2. The testcase tests/kernel/mem_protect/mem_protect/src/main.c also needs fixing.

nashif, 2018-04-30:
both issues now fixed.

closed 2018-04-30 by nashif