zephyrproject-rtos/zephyr #7213: DTS should prefix all preprocessor macros with CONFIG_

SebastianBoe, 2018-04-26:
Kconfig and DTS are inconsistent in their naming convention. In Kconfig all defines are prefixed with CONFIG_. In DTS it is a mix of CONFIG_ and no prefix. Unless there is a good reason for DTS to generate macros without the CONFIG_ prefix we should start enforcing that DTS limits itself to CONFIG_.

Consistently prefixing with CONFIG_ effectively creates a namespace 
and that is important for two reasons:

It reduces the likelihood of encountering aliasing issues with application code
if we can have a simple rule like:

"Preprocessor macros prefixed with CONFIG_ are reserved by the configuration system (DTS+Kconfig)."

It also makes it more predictable to a user where macros come from and therefore makes it easier to read the code, e.g. if he sees a macro:

FLASH_NRF_FLASH_WRITE_BLOCK_SIZE

vs.

CONFIG_FLASH_NRF_FLASH_WRITE_BLOCK_SIZE

erwango, 2018-04-27:
@SebastianBoe , do you mean that everything which is generated from dts should use CONFIG_ name space? 
Such as NORDIC_NRF5_I2C_40003000_LABEL would be CONFIG_NORDIC_NRF5_I2C_40003000_LABEL ?

SebastianBoe, 2018-04-27:
Yes. Specifically everything in generated_dts_board.h.

erwango, 2018-04-27:
Not sure to agree in that case. 

Having dts generated bit and Kcong using same namespace would create confusion as we're not able anymore to say if this should be configured from Kcong or from dts.
Also I find it would be confusing to see, in .fixup files (even if we should eventually get rid of those):
#define CONFIG_UART_STM32_PORT_1_NAME	  CONFIG_ST_STM32_USART_40013800_LABEL

Besides, eveything dts generated is not available from Kconfig anymore. So these are different set of preprocessor macros. Eventually we should have a DTS_ namespace.

Last point, having no name space today allows for direct generation of aliases (cf #6979), and hence to get rid of bindings (such as .fixup). Having namespace for generated bits would enforce the need for bindings files. If required, we can live with it, but it's nice to live w/o it.

SebastianBoe, 2018-04-27:
> Having dts generated bit and Kcong using same namespace would create confusion as we're not able anymore to say if this should be configured from Kcong or from dts.

Not sure how that is better than today's situation, today there is a mix of CONFIG_ and non-CONFIG_ in DTS so you are not able to predict this now either.

> Also I find it would be confusing to see, in .fixup files (even if we should eventually get rid of those):
> #define CONFIG_UART_STM32_PORT_1_NAME	CONFIG_ST_STM32_USART_40013800_LABEL

Could you elaborate on what is confusing?

> Besides, eveything dts generated is not available from Kconfig anymore. So these are different set of preprocessor macros.

Correct, they are not available at the same time. But IMHO they should be available simultaneously as explained in issue #7214 .

> Eventually we should have a DTS_ namespace.

I don't see the need for the configuration system to use a separate namespace for DTS_, but that would at least be better than todays situation.

[0] https://github.com/zephyrproject-rtos/zephyr/issues/7214

b0661, 2018-04-29:
The meaning of CONFIG must be clear and unambiguous to the user. It shall also be clear who is the user.

In my view the
- user is the application developer (vs. driver/ kernel developer) and 
- the meaning of CONFIG is "the application developer does have a choice".

Within this semantic system FLASH_NRF_FLASH_WRITE_BLOCK_SIZE should be FLASH_NRF_FLASH_WRITE_BLOCK_SIZE or DTS_FLASH_NRF_FLASH_WRITE_BLOCK_SIZE  as there is most probably no choice (write block size is a property of the HW and not dependent on or controllable by the application).

SebastianBoe, 2018-04-30:
@b0661 , are you proposing prefixing DTS defines with DTS_ or on a case-by-case basis either prefixing DTS defines with DTS_ or CONFIG_ depending on whether they should be application-controllable or not?

Technically everything in DTS is (and needs to be) controllable-by-application (unless we start limiting DTS to describe HW). Practically however it is not obvious what values are controllable and which are not. Some defines that one would assume were non-controllable, such as flash size, are commonly (unfortunately) overridden by application's to make space for their own modules.

b0661, 2018-04-30:
@SebastianBoe, I think before investing the effort of re-organizing and prefixing a value by CONFIG (or any other "class" prefix) the meaning of such a prefix and the usage shall be clear and finally also documented. 

At the moment the meaning of CONFIG is unclear. Mostly it may be "configuration variable created by kconfig" , but also others are using the CONFIG prefix. The perspective is tool oriented and not usage oriented. This may be fuilly acceptable if the goal is to identify the tool that created the configuration variable. This is also the most easy convention to impose. Another meaning, of CONFIG that I think you proposed is "this is a configuration variable of Zephyr" where you can combine DTS and kconfig.

My proposal is to look from the application developer view on Zephyr. I did a choice by kconfig and dts. CONFIG_xx documents my choice. Based on the CONFIG_xx defines I can adapt my application. The primary user of the CONFIG_xx defines is the application (kernel/ drivers still aso have to adhere to it). 

> Technically everything in DTS is (and needs to be) controllable-by-application (unless we start limiting DTS to describe HW).

I agree technically an application can alter everything in a DTS. To be more precise CONFIG should be used for the properties that are foreseen to be altered by the application - e.g.:
- activation of a driver (status)
  This is implicit as defines are only generated if status="ok".
- default clock selection/ enable/ divider/ multiplier ( assigned-clocks,  assigned-clock-parents, assigned-clock-rates)
- default pin selection/ mux/ control  (pinctrl-x with pinctrl-name(x) = "default")
- label (for this I´m not really convinced)

>are you proposing prefixing DTS defines with DTS_ or on a case-by-case basis either prefixing DTS defines with DTS_ or CONFIG_ depending on whether they should be application-controllable or not?

To follow the "CONFIG_xx documents the application developwer choice" the CONFIG_xx defines may just be aliases. This way the defines generation for DTS can be kept uniform.  The answer to your question is "both" !-). DTS is for driver/kernel developers to make clear where the define is coming from. The CONFIG aliases are for the application developer. 

> Some defines that one would assume were non-controllable, such as flash size, are commonly (unfortunately) overridden by application's to make space for their own modules.

Such defines should be split up in a total flash size (which is an inherent property of the HW) and one CONFIG_xx define for the flash space an application want´s to allocate. This might even help to detect allocation problems more clearly.

SebastianBoe, 2018-05-02:
> To follow the "CONFIG_xx documents the application developwer choice" the CONFIG_xx defines may just be aliases. This way the defines generation for DTS can be kept uniform. The answer to your [...]

This system would satisfy the main problem that this issue was trying to address, will re-write this issue to not include a solution, but just a problem description.

The complexity of this all annoys me though. I wonder if it would be possible to simplify the configuration system such that DTS_ is inherently not an application developer choice and CONFIG_ is. E.g. by limiting DTS_ to describe HW and not configure HW.

erwango, 2018-05-02:
> At the moment the meaning of CONFIG is unclear.

@SebastianBoe , @b0661 , I'm trying to link this conversation with issue #7159 fixed by https://github.com/zephyrproject-rtos/zephyr/commit/63cb2334f07006f6b89fbdeadebbff2445b3df90 
(tell me if I'm off topic).

Visible and invisible Kconfig symbols are there defined, and from what I understand:
- only invisible symbols (board dependant) should be defined in boards Kconfig.defconfig files (which by the way involves that all Kconfig symbols should be reviewed vs this criteria, which was omitted in the PR). Invisible symbols don't feature prompt in their initial Kconfig file and then can't be accessed/modified by menuconfig
- visible symbols feature a prompt, and then can be modified in menuconfig. These symbols are then application dependant.

@b0661, in the system you described, I guess the CONFIG_ that would alias DTS_ are the visible ones. Am I correct ?
For instance device tree specifies a clock setting, which I am able to modify via menuconfig. While CONFIG_CLOCK_CONTROL_STM32_CUBE is selected because I'm running application on a STM32 SoC and hence does not get to be modified.  
Hence, all the CONFIG_ flags set to day in boards Kconfig.defconfig that are used today to provide a default application setting per board (like I2C instance, serial port activated if console is defined, default network interface ...) will be replaced by device tree.

This is coherent, and calls for a global review of Kconfig flags definition:
- Provide a prompt for Kconfig symbols which will eventually get an equivalent generated DTS_ defines (these are alias CONFIG_ symbols)
- Remove prompt for the Kconfig symbols that are not application dependant (ie make them invisible)

Am I being too simplistic here?

SebastianBoe, 2018-05-02:
> Hence, all the CONFIG_ flags set to day in boards Kconfig.defconfig that are used today to provide a default application setting per board (like I2C instance, serial port activated if console is defined, default network interface ...) will be replaced by device tree.
-- @erwango 

I agree that Kconfig.defconfig should be deprecated and it's metadata should be moved into DeviceTree.

> Remove prompt for the Kconfig symbols that are not application dependant (ie make them invisible)
-- @erwango 

I also agree that options that are not application-configurable should not be displayed as configuration options in the menu tree.

> Provide a prompt for Kconfig symbols which will eventually get an equivalent generated DTS_ defines (these are alias CONFIG_ symbols)
-- @erwango 

Not sure what you mean. But we can't have a Kconfig option that is not set appropriately. The natural solution then is to have them set according to DTS information, as #7302 proposes.

erwango, 2018-05-02:
> I agree that Kconfig.defconfig should be deprecated and it's metadata should be moved into DeviceTree

I meant it only for "visible symbols", according to https://github.com/zephyrproject-rtos/zephyr/commit/63cb2334f07006f6b89fbdeadebbff2445b3df90, Kconfig.defconfig is usefull for "invisible symbols"

 >>   Provide a prompt for Kconfig symbols which will eventually get an equivalent generated DTS_ defines (these are alias CONFIG_ symbols)
    -- @erwango

> Not sure what you mean. But we can't have a Kconfig option that is not set appropriately. The natural solution then is to have them set according to DTS information, as #7302 proposes.

I mean that, in the end (when we have achieved to generate everything we need from device tree), Kconfig symbols with a prompt are aliases to DTS_ symbols (IIUC what was proposed by @b0661).

erwango, 2019-02-15:
@SebastianBoe , do you think this one can be closed now?

SebastianBoe, 2019-02-15:
No, there are still symbols in generated_dts_board_unfixed.h that don't have the prefix DT_

galak, 2019-04-30:
> No, there are still symbols in generated_dts_board_unfixed.h that don't have the prefix DT_

There are a few special cases for the samples, we should decide what we want to do there, otherwise I think everything else has a DT_ prefix.

galak, 2019-07-30:
I think this can be closed now.  Everything is DT_ prefixed or marked deprecated now.

SebastianBoe, 2019-07-30:
I'd prefer to close it when there are no more deprecated symbols being generated.

galak, 2019-07-31:
> I'd prefer to close it when there are no more deprecated symbols being generated.

works for me.

galak, 2019-07-31:
NOTE: This issue is practically closed, but we can officially close it once we remove the old deprecated defines that we are generating in what will be release 2.0.0

pabigot, 2019-09-06:
AFAICT this is complete: code in master no longer generates deprecated symbols.

closed 2019-09-06 by pabigot