zephyrproject-rtos/zephyr #7214: Defines from DTS and Kconfig should be available simultaneously

SebastianBoe, 2018-04-26:
Currently Kconfig defines are available in all C sources through gcc's "-include file.h" mechanism whereas dts defines must be explicitly included through the preprocessor's "#include file.h" directive.

This means that firmware must treat configuration coming from DTS and Kconfig differently. This difference makes the configuration system harder to understand and can cause users to be confused about why a CONFIG_ from DTS is not available in his FW, but a CONFIG_ from Kconfig is.

It would be better to use the same mechanism and thereby ensure that FW can be agnostic to what mechanism was used to create the config.

galak, 2019-02-09:
@SebastianBoe is this still an issue?

SebastianBoe, 2019-02-11:
Yes. We should use the same mechanism to enter defines from DT and Kconfig into C source code.

mbolivar-nordic, 2021-05-26:
I'm taking the liberty of closing this. I don't think it's that confusing in practice.

closed 2021-05-26 by mbolivar-nordic