zephyrproject-rtos/zephyr #7227: /subsys/storage/flash_map/flash_map_default.c missing declarations.

leodesigner, 2018-04-27:
Hi,
Testing settings storage subsystem.
Got missed declarations for:
```
zephyr-current/subsys/storage/flash_map/flash_map_default.c:49:13: error: ‘FLASH_AREA_STORAGE_OFFSET’ undeclared here (not in a function)
   .fa_off = FLASH_AREA_STORAGE_OFFSET,
             ^~~~~~~~~~~~~~~~~~~~~~~~~
zephyr-current/subsys/storage/flash_map/flash_map_default.c:50:14: error: ‘FLASH_AREA_STORAGE_SIZE’ undeclared here (not in a function)
   .fa_size = FLASH_AREA_STORAGE_SIZE,
              ^~~~~~~~~~~~~~~~~~~~~~~
```
I have defined CONFIG_FS_FLASH_MAP_STORAGE
and CONFIG_SETTINGS_FCB_FLASH_AREA=4
target SOC is CONFIG_SOC_SERIES_NRF52X

should I use CONFIG_SETTINGS_FCB_FLASH_AREA=3 for now and disable CONFIG_FS_FLASH_MAP_STORAGE ?

Thanks.

leodesigner, 2018-04-27:
Storage definition should be added to DTS for every board.
Made a temporary fix for nrf52_blenano2
in boards/arm/nrf52_blenano2/nrf52_blenano2.dts
```
                storage_partition: partition@7d000 {
                        label = "storage";
                        reg = <0x0007d000 0x4000>;
                };
```
Now the missed definition is present.
I am not familiar with the DTS tree system, is this change should be added to every board ? Or there is some kind of template for these configs ?

nvlsianpu, 2018-05-04:
This is not a bug - just as @leodesigner found, declaration of storage space is undefined for this board. For instance it is possible to add such by DT overlay file. Anyway I open PR for improve this.

leodesigner, 2018-05-05:
Thanks @nvlsianpu, I was hoping it could be done for every board all at once.
But anyway, closing the issue for now.

closed 2018-05-05 by leodesigner