zephyrproject-rtos/zephyr #7230: The guidelines for whether something should be in DTS or Kconfig are too vague

SebastianBoe, 2018-04-27:
We need improved guidelines for determining if a configuration knob belongs in Kconfig or DTS.
The existing guidelines are attached below.

From the guidelines one can read that DTS should be used for:

- HW description
- Zephyr-specific configuration
- Some of the use-cases where Kconfig is currently applied
- Not all of the use-cases where CMSIS (and similair) is currently applied

This is pretty vague in my opinion.

#### Zephyr DTS guidelines
> In Zephyr, device tree is used to not only describe hardware, but also to describe Zephyr-specific configuration information. The Zephyr-specific information is intended to augment the device tree descriptions. The main reason for this is to leverage existing device tree files that a SoC vendor may already have defined for a given platform.
> 
> Today, configuration in Zephyr comes from a number of different places. It can come from Kconfig files, CMSIS header files, vendor header files, prj.conf files, and other miscellaneous sources. The intent of using device tree is to replace or curtail the use of Kconfig files throughout the system, and instead use device tree files to describe the configuration of device nodes. CMSIS and vendor header files can be used in conjunction with the device tree to fully describe hardware. Device tree is not intended to replace CMSIS or vendor include files.

carlescufi, 2018-05-07:
Assigning to @galak since I believe he's in the better position to evaluate how to approach this.

erwango, 2018-05-31:
As commented in #7302, DTS has (/is) a specification (https://www.devicetree.org/), so I think we should rely on it

PR #8507 (2018-06-25): doc: dt: Clarify the relationship between DT and Kconfig
Try to clarify the scope of Device Tree vs Kconfig in order to allow
Zephyr users to identify where a particular configuration option when
developing an application, board support or SoC support.

Fixes #7230
commit 45745052 (2018-06-25): doc: dt: Clarify the relationship between DT and Kconfig

Try to clarify the scope of Device Tree vs Kconfig in order to allow
Zephyr users to identify where a particular configuration option when
developing an application, board support or SoC support.

Fixes #7230
closed 2018-06-25 by carlescufi