zephyrproject-rtos/zephyr #7243: BLE DTM ll_test does not set correct TXPower

paulgavrikov, 2018-04-28:
According to the Bluetooth Core Spec v5 Direct Test Mode shall transmit with maximum txpower. However ll_test sets the txpower to 0dBm, which is not the highest txpower for many device including Nordics nRF52 family.

cvinayak, 2018-05-02:
@paulgavrikov if you do have a PR to use the Max Tx Power for nRF5 series SoCs, I will be happy to review the PR.
@carlescufi This will need use re-run tests on the RF tester.

paulgavrikov, 2018-05-15:
@cvinayak I don't think I can create a PR for this repo. But you can checkout [this commit ](https://github.com/zephyrproject-rtos/zephyr/commit/a41f4a7b0cc214daaa1eb20953447900f907e766)

cvinayak, 2018-06-02:
@paulgavrikov ok. I will prepare a PR and let you know soon.

PR #8122 (2018-06-12): Bluetooth: controller: Fix to use max. tx power in DTM test mode
According to the Bluetooth Specification v5.0, Direct Test
Mode shall use maximum Tx power.

Fixed by adding nRF5x Radio HAL interfaces to get supported
maximum Tx power for SoC selected, and DTM testing sets the
Radio peripheral to use maximum Tx power.

Fixes #7243
commit 362a6b34 (2018-06-12): Bluetooth: controller: Fix to use max. tx power in DTM test mode

According to the Bluetooth Specification v5.0, Direct Test
Mode shall use maximum Tx power.

Fixed by adding nRF5x Radio HAL interfaces to get supported
maximum Tx power for SoC selected, and DTM testing sets the
Radio peripheral to use maximum Tx power.

Fixes #7243
closed 2018-06-12 by carlescufi