zephyrproject-rtos/zephyr #7245: EMSK 7d: can't build after 60ec8be309cb84d72c5fc61330abc968eb62333e

inakypg, 2018-04-29:
Hello 

Starting said commit

```
    arch: em{7,9,11}d: Add dts support
    
    Add initial device tree support for the em{7,9,11}d SoC and associated
    em_starterkit boards.  The device tree at this point specifies cpu core,
    memory, interrupt controller, uart's and i2c controllers.
    
    Signed-off-by: Kumar Gala <kumar.gala@linaro.org>
```

all builds in our HW automation for EMSK 7d model are failing with:

```
...
[ 93%] Linking C executable zephyr_prebuilt.elf
/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/../../libexec/arc-zephyr-elf/gcc/arc-zephyr-elf/6.2.1/real-ld: invalid origin for memory region SRAM
collect2: error: ld returned 1 exit status
...
```

nashif, 2018-04-29:
How do you build and what board definition are you using exactly? This builds just fine in sanitycheck and locally..

inakypg, 2018-04-29:
Hmm, I thought I had added this part, but I guess I missed it:

Add a configuration file that enables:

```
CONFIG_SOC_EM7D=y
CONFIG_SOC_EM9D=n
CONFIG_SOC_EM11D=n
```

to make sure oyu are building 7D

nashif, 2018-04-29:
I have that, you need to show this can be reproduced with a sample, for example:

```
cd samples/hello_world
mkdir build; cd build
cmake -DBOARD=em_starterkit_em7d ..
make
```

inakypg, 2018-04-29:
```
$ set | grep -i ZEPHYR
ZEPHYR_BASE=/home/inaky/z/kernel.git
ZEPHYR_GCC_VARIANT=zephyr
ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk-0.9.2
ZEPHYR_TOOLCHAIN_VARIANT=zephyr
$ git -C $ZEPHYR_BASE describe
v1.11.0-875-gd9cdf7bf30
$ rm -rf build
$ mkdir build
$ cat > build/500_emsk_config.conf <<EOF
CONFIG_SOC_EM7D=y
CONFIG_SOC_EM9D=n
CONFIG_SOC_EM11D=n
EOF
$ cmake -DBOARD=em_starterkit -Bbuild -Hzephyr.git/samples/hello_world/
CMake Deprecation Warning at /home/inaky/z/kernel.git/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.4", minimum required is "3.4") 
-- Selected BOARD em_starterkit
Zephyr version: 1.11.99
Parsing Kconfig tree in /home/inaky/z/kernel.git/Kconfig
Using /home/inaky/z/kernel.git/boards/arc/em_starterkit/em_starterkit_defconfig as base
Merging /home/inaky/z/s/production/zephyr.git/samples/hello_world/prj.conf
Merging /home/inaky/z/s/production/build/500_emsk_config.conf
/home/inaky/z/s/production/build/500_emsk_config.conf:2: warning: SOC_EM9D (defined at /home/inaky/z/kernel.git/arch/arc/soc/em9d/Kconfig.soc:2) set more than once. Old value: "y", new value: "n".
CMake Warning at /home/inaky/z/kernel.git/cmake/toolchain.cmake:13 (message):
  ZEPHYR_GCC_VARIANT is deprecated, please use ZEPHYR_TOOLCHAIN_VARIANT
  instead
Call Stack (most recent call first):
  /home/inaky/z/kernel.git/cmake/app/boilerplate.cmake:243 (include)
  CMakeLists.txt:2 (include)

-- Generating zephyr/include/generated/generated_dts_board.h
-- Cache files will be written to: /home/inaky/.cache/zephyr
-- The C compiler identification is GNU 6.2.1
-- The CXX compiler identification is GNU 6.2.1
-- The ASM compiler identification is GNU
-- Found assembler: /opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/arc-zephyr-elf-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/inaky/z/s/production/build
$ make -C build
make: Entering directory '/home/inaky/z/s/production/build'
make[1]: Entering directory '/home/inaky/z/s/production/build'
...
[ 94%] Linking C executable zephyr_prebuilt.elf
/opt/zephyr-sdk-0.9.2/sysroots/x86_64-pokysdk-linux/usr/bin/arc-zephyr-elf/../../libexec/arc-zephyr-elf/gcc/arc-zephyr-elf/6.2.1/real-ld: invalid origin for memory region SRAM
collect2: error: ld returned 1 exit status
make[2]: *** [zephyr/CMakeFiles/zephyr_prebuilt.dir/build.make:101: zephyr/zephyr_prebuilt.elf] Error 1
make[2]: Leaving directory '/home/inaky/z/s/production/build'
make[1]: *** [CMakeFiles/Makefile2:538: zephyr/CMakeFiles/zephyr_prebuilt.dir/all] Error 2
make[1]: Leaving directory '/home/inaky/z/s/production/build'
make: *** [Makefile:84: all] Error 2
make: Leaving directory '/home/inaky/z/s/production/build'
```

inakypg, 2018-04-29:
There is no BOARD=em_starterkit_7d or whatever, only one board that then is multiplexed (the wrong way, if you ask me).

nashif, 2018-04-30:
why are you doing this:

```
$ cat > build/500_emsk_config.conf <<EOF
CONFIG_SOC_EM7D=y
CONFIG_SOC_EM9D=n
CONFIG_SOC_EM11D=n
EOF
```

em_starterkit is already configured as em9d, if you want em7d, then just use 

`cmake -DBOARD=em_starterkit_em7d`

What you end up with is an invalid configuration mixing incompatible features, the difference:

```
i9:hello_world(master): diff -u  /tmp/anas.conf /tmp/inaky.conf
--- /tmp/anas.conf	2018-04-29 20:47:58.104570632 -0500
+++ /tmp/inaky.conf	2018-04-29 20:47:10.183151752 -0500
@@ -11,8 +11,6 @@
 # General Architecture Options
 #
 # CONFIG_HW_STACK_PROTECTION is not set
-# CONFIG_USERSPACE is not set
-CONFIG_PRIVILEGED_STACK_SIZE=256
 # CONFIG_STACK_GROWS_UP is not set
 # CONFIG_SIMPLE_FATAL_ERROR_HANDLER is not set

@@ -25,7 +23,6 @@
 CONFIG_GEN_IRQ_START_VECTOR=16
 # CONFIG_IRQ_OFFLOAD is not set
 CONFIG_ARCH_HAS_STACK_PROTECTION=y
-CONFIG_ARCH_HAS_USERSPACE=y
 CONFIG_CPU_HAS_MPU=y
 CONFIG_ARCH="arc"
 CONFIG_SOC="em7d"
@@ -63,11 +60,10 @@
 #
 # ARC MPU Options
 #
-CONFIG_ARC_MPU_ENABLE=y
+# CONFIG_ARC_MPU_ENABLE is not set
 CONFIG_ARC_MPU_VER=3
-CONFIG_ARC_CORE_MPU=y
-# CONFIG_MPU_STACK_GUARD is not set
-CONFIG_ARC_MPU=y
+# CONFIG_ARC_CORE_MPU is not set
+# CONFIG_ARC_MPU is not set
 # CONFIG_CACHE_LINE_SIZE_DETECT is not set
 CONFIG_CACHE_LINE_SIZE=32
 CONFIG_CACHE_FLUSHING=y
@@ -92,7 +88,7 @@
 CONFIG_MAIN_STACK_SIZE=1024
 CONFIG_IDLE_STACK_SIZE=320
 CONFIG_ISR_STACK_SIZE=2048
-CONFIG_THREAD_STACK_INFO=y
+# CONFIG_THREAD_STACK_INFO is not set
 # CONFIG_THREAD_CUSTOM_DATA is not set
 CONFIG_ERRNO=y
 # CONFIG_APPLICATION_MEMORY is not set
@@ -445,4 +441,3 @@
 # CONFIG_ZTEST is not set
 # CONFIG_TEST is not set
 CONFIG_TEST_EXTRA_STACKSIZE=0
-# CONFIG_TEST_USERSPACE is not set
```

nashif, 2018-04-30:
> There is no BOARD=em_starterkit_7d or whatever, only one board that then is multiplexed (the wrong way, if you ask me).

there is a em_starterkit_em7d configuration. the multiplexing is being fixed in #7238

nashif, 2018-04-30:
you need to check what firmware you have installed, for 2.2, you need to use `em_starterkit_em7d_v22`

inakypg, 2018-05-01:
Initial test reports hello_world working using said board and

```
$ git log -10
b12b63f9da (HEAD) dts: fixes in the dts of em_starterkit em9d configuration
315ad71253 doc: update the doc of em_starterkit
fbd6a6cf49 dts: optimize the dts for em_starterkit
ec0bcb8d07 arch: arc: refactor the soc part of em_starterkit
f59a68b6d8 samples: net: http: Add TLS compilation test...
```

I will need to update the HW automation environment to report a different board -- thanks for the update.

inakypg, 2018-05-03:
Verified working as expected, so closing as not a bug

closed 2018-05-03 by inakypg