zephyrproject-rtos/zephyr #7246: esp32 fails to build with xtensa-esp32-elf-gcc: error: unrecognized command line option '-no-pie'

inakypg, 2018-04-29:
bisect says after commit:
```
commit 71b849f18c6af931a004754b3ebf76a5c1dfa715

    cmake: Port Zephyr to use zephyr_check_compiler_flag
    
    This commit ports nearly all usage of check_c_compiler_flag,
    check_cxx_compiler_flag, and check_compiler_flag to use
    zephyr_check_compiler_flag instead.
    
    This has a significant CMake configure-time performance impact.
    
    Signed-off-by: Sebastian Bøe <sebastian.boe@nordicsemi.no>

```
_hello world_  (and multiple others) starting failing to build with ZEPHYR_TOOLCHAIN_VARIANT=espressif (v5.2.0), ESP_IDF rev v3.0-dev-388-gdc8c3389

```
...
[ 89%] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[ 89%] Built target zephyr
[ 90%] Linking C static library libkernel.a
[ 90%] Built target kernel
Scanning dependencies of target zephyr_prebuilt
[ 92%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[ 93%] Linking C executable zephyr_prebuilt.elf
xtensa-esp32-elf-gcc: error: unrecognized command line option '-no-pie'
```

inakypg, 2018-04-30:
doesn't work for me:

```
[inaky@inaky-mobl production]$ git -C zephyr.git/ log -10
59e59e0585 (HEAD -> SebastianBoe-fix_check_compiler_flag) cmake: Suppress messages about compiler checks
e4c48b2208 cmake: check_compiler_flag: Fix bug where checks were aliased
d9cdf7bf30 (master, refs/bisect/bad) arch/arm/soc/st_stm32: Fix typos in soc.h
517a08ee13 (refs/bisect/good-517a08ee1370cf3ae42557347d1f6a4e386a2261) CODEOWNERS: Add myself as the Codeowner for LED API and drivers
42a7c5732f arm: soc: Elaborate Kconfig strings for MPU selection
c36e5ac129 (refs/bisect/good-c36e5ac129b26c5ae12b8f39496fcaabc209a049) arm: Fix title for SoC configuration in Kconfig
b27d96850f samples: bluetooth: Add Apple iBeacon demo application
034a11b7cd boards: arm: 96b_neonkey: Fix I2C_3 for LP3943 LED driver
1572594b53 (refs/bisect/good-1572594b532ea36f81661bab76231d3d3cf91cbb) samples: drivers: Add sample application for LP3943
bb394bbafb drivers: led: Add LED driver support for TI LP3943
[inaky@inaky-mobl production]$ set | grep -i ZEPHYR
OLDPWD=/home/inaky/z/s/production/zephyr.git
ZEPHYR_BASE=/home/inaky/z/kernel.git
ZEPHYR_SDK_INSTALL_DIR=/opt/zephyr-sdk-0.9.2
[inaky@inaky-mobl production]$ rm -rf build/; mkdir build
[inaky@inaky-mobl production]$ cmake -DZEPHYR_TOOLCHAIN_VARIANT=espressif -DBOARD=esp32 -Bbuild -Hzephyr.git/samples/hello_world/
CMake Deprecation Warning at /home/inaky/z/kernel.git/cmake/app/boilerplate.cmake:38 (cmake_policy):
  The OLD behavior for policy CMP0000 will be removed from a future version
  of CMake.

  The cmake-policies(7) manual explains that the OLD behaviors of all
  policies are deprecated and that a policy should be set to OLD only under
  specific short-term circumstances.  Projects should be ported to the NEW
  behavior and not rely on setting a policy to OLD.
Call Stack (most recent call first):
  CMakeLists.txt:2 (include)

-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.6.4", minimum required is "3.4") 
-- Selected BOARD esp32
Zephyr version: 1.11.99
Parsing Kconfig tree in /home/inaky/z/kernel.git/Kconfig
Using /home/inaky/z/kernel.git/boards/xtensa/esp32/esp32_defconfig as base
Merging /home/inaky/z/s/production/zephyr.git/samples/hello_world/prj.conf
warning: ENTROPY_ESP32_RNG (defined at /home/inaky/z/kernel.git/drivers/entropy/Kconfig.esp32:7) was assigned the value "y" but got the value "n" -- check dependencies
-- Generating zephyr/include/generated/generated_dts_board.h
-- Cache files will be written to: /home/inaky/.cache/zephyr
CMake Warning at /home/inaky/z/kernel.git/boards/common/esp32.board.cmake:5 (message):
  Setting ESP_IDF_PATH in the environment is deprecated.  Use cmake
  -DESP_IDF_PATH=...  instead.
Call Stack (most recent call first):
  /home/inaky/z/kernel.git/boards/xtensa/esp32/board.cmake:1 (include)
  /home/inaky/z/kernel.git/cmake/app/boilerplate.cmake:284 (include)
  CMakeLists.txt:2 (include)

-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /opt/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/inaky/z/s/production/build
[inaky@inaky-mobl production]$ make -C build
make: Entering directory '/home/inaky/z/s/production/build'
make[1]: Entering directory '/home/inaky/z/s/production/build'
make[2]: Entering directory '/home/inaky/z/s/production/build'
Scanning dependencies of target kobj_types_h_target
make[2]: Leaving directory '/home/inaky/z/s/production/build'
make[2]: Entering directory '/home/inaky/z/s/production/build'
[  1%] Generating include/generated/kobj-types-enum.h, include/generated/otype-to-str.h
make[2]: Leaving directory '/home/inaky/z/s/production/build'
[  1%] Built target kobj_types_h_target
make[2]: Entering directory '/home/inaky/z/s/production/build'
Scanning dependencies of target syscall_macros_h_target
make[2]: Leaving directory '/home/inaky/z/s/production/build'
[... 103 lines cut ...]
[ 72%] Building ASM object zephyr/CMakeFiles/zephyr.dir/arch/xtensa/core/atomic.S.obj
[ 73%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/crc/crc16_sw.c.obj
[ 75%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/crc/crc8_sw.c.obj
[ 76%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/mempool/mempool.c.obj
[ 77%] Building C object zephyr/CMakeFiles/zephyr.dir/lib/rbtree/rb.c.obj
[ 78%] Building C object zephyr/CMakeFiles/zephyr.dir/misc/printk.c.obj
[ 80%] Building C object zephyr/CMakeFiles/zephyr.dir/misc/generated/configs.c.obj
[ 81%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/console/uart_console.c.obj
[ 82%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/gpio/gpio_esp32.c.obj
[ 84%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/i2c/i2c_esp32.c.obj
[ 85%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/pinmux/pinmux_esp32.c.obj
[ 86%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/serial/uart_esp32.c.obj
[ 88%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/timer/sys_clock_init.c.obj
[ 89%] Building C object zephyr/CMakeFiles/zephyr.dir/drivers/timer/xtensa_sys_timer.c.obj
[ 90%] Linking C static library libzephyr.a
make[2]: Leaving directory '/home/inaky/z/s/production/build'
[ 90%] Built target zephyr
make[2]: Entering directory '/home/inaky/z/s/production/build'
Scanning dependencies of target zephyr_prebuilt
make[2]: Leaving directory '/home/inaky/z/s/production/build'
make[2]: Entering directory '/home/inaky/z/s/production/build'
[ 92%] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[ 93%] Linking C executable zephyr_prebuilt.elf
xtensa-esp32-elf-gcc: error: unrecognized command line option '-no-pie'
```

inakypg, 2018-04-30:
same xtensa SDK version as in the bug report

nashif, 2018-05-02:
actually this is failing with other toolchains as well, i.e. issm toolchain and xcc.

SebastianBoe, 2018-05-02:
"doesn't work for me:"

Strange. It works on my machine.

Could you try clearing the cache?
Also, make sure you have a clean git status and clean build directory.

rm -rf /home/inaky/.cache/zephyr

```
sebo@mach:~/zephyr/samples/hello_world$ /home/sebo/workspace_zephyr/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc --version
xtensa-esp32-elf-gcc (crosstool-NG crosstool-ng-1.22.0-61-gab8375a) 5.2.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
sebo@mach:~/zephyr/samples/hello_world$ git log -1
WARNING: terminal is not fully functional
-  (press RETURN)
commit 59e59e05853f4304349cb9c2dfd4845773ed92d7 (HEAD -> fix_check_compiler_flag, sebo_remote/fix_check_compiler_flag)
Author: Sebastian Bøe <sebastian.boe@nordicsemi.no>
Date:   Mon Apr 30 10:59:56 2018 +0200

    cmake: Suppress messages about compiler checks
    
    Messages about compiler checks are very verbose so we disable them by
    default. They have always been undesirably verbose, but introducing
    CMAKE_REQUIRED_FLAGS into the 'check' string made them unacceptably
    so.
    
    Signed-off-by: Sebastian Bøe <sebastian.boe@nordicsemi.no>
sebo@mach:~/zephyr/samples/hello_world$ source ~/.zephyr_esp.sh 
sebo@mach:~/zephyr/samples/hello_world$ cat ~/.zephyr_esp.sh 
export ESPRESSIF_TOOLCHAIN_PATH="$HOME/workspace_zephyr/esp/xtensa-esp32-elf"
export ESP_IDF_PATH="$HOME/workspace_zephyr/esp/esp-idf"
export ZEPHYR_TOOLCHAIN_VARIANT="espressif"

export PATH="$PATH:$ESPRESSIF_TOOLCHAIN_PATH/bin"
export IDF_PATH="$ESP_IDF_PATH"
sebo@mach:~/zephyr/samples/hello_world$ rm -rf build && cmake -DZEPHYR_TOOLCHAIN_VARIANT=espressif -DBOARD=esp32 -Bbuild -H${ZEPHYR_BASE}/samples/hello_world/
-- Found PythonInterp: /usr/bin/python3 (found suitable version "3.5.2", minimum required is "3.4") 
-- Selected BOARD esp32
Zephyr version: 1.11.99
Parsing Kconfig tree in /home/sebo/zephyr/Kconfig
Using /home/sebo/zephyr/boards/xtensa/esp32/esp32_defconfig as base
Merging /home/sebo/zephyr/samples/hello_world/prj.conf
warning: ENTROPY_ESP32_RNG (defined at /home/sebo/zephyr/drivers/entropy/Kconfig.esp32:7) was assigned the value "y" but got the value "n" -- check dependencies
-- Generating zephyr/include/generated/generated_dts_board.h
-- Cache files will be written to: /home/sebo/.cache/zephyr
CMake Warning at /home/sebo/zephyr/boards/common/esp32.board.cmake:5 (message):
  Setting ESP_IDF_PATH in the environment is deprecated.  Use cmake
  -DESP_IDF_PATH=...  instead.
Call Stack (most recent call first):
  /home/sebo/zephyr/boards/xtensa/esp32/board.cmake:1 (include)
  /home/sebo/zephyr/cmake/app/boilerplate.cmake:284 (include)
  CMakeLists.txt:2 (include)

-- The C compiler identification is GNU 5.2.0
-- The CXX compiler identification is GNU 5.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/sebo/workspace_zephyr/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Performing Test toolchain_is_ok
-- Performing Test toolchain_is_ok - Success
-- Configuring done
-- Generating done
-- Build files have been written to: /home/sebo/zephyr/samples/hello_world/build
sebo@mach:~/zephyr/samples/hello_world$ ninja -C build/
ninja: Entering directory `build/'
[1/76] Generating always_rebuild
Building for board esp32
[2/76] Generating include/generated/syscall_macros.h
[... 50 lines cut ...]
[53/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/msg_q.c.obj
[54/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mailbox.c.obj
[55/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/mempool.c.obj
[56/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/stack.c.obj
[57/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/version.c.obj
[58/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/queue.c.obj
[59/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread_abort.c.obj
[60/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sem.c.obj
[61/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/smp.c.obj
[62/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/system_work_q.c.obj
[63/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sys_clock.c.obj
[64/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/work_q.c.obj
[65/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/pipes.c.obj
[66/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/sched.c.obj
[67/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/thread.c.obj
[68/76] Building C object zephyr/kernel/CMakeFiles/kernel.dir/timer.c.obj
[69/76] Linking C static library zephyr/kernel/libkernel.a
[70/76] Building C object zephyr/CMakeFiles/zephyr_prebuilt.dir/misc/empty_file.c.obj
[71/76] Linking C executable zephyr/zephyr_prebuilt.elf
[72/76] Generating linker_pass_final.cmd
[73/76] Generating isr_tables.c
[74/76] Building C object zephyr/CMakeFiles/kernel_elf.dir/misc/empty_file.c.obj
[75/76] Building C object zephyr/CMakeFiles/kernel_elf.dir/isr_tables.c.obj
[76/76] Linking C executable zephyr/zephyr.elf
```

SebastianBoe, 2018-05-02:
@inakypg : Please dump the cache file (after clearing it and re-running CMake).

And also place

print(key)

in cmake/extensions.cmake/zephyr_check_compiler_flag

and a

message("Cache miss")

at the end of the function.

closed 2018-05-02 by nashif

inakypg, 2018-05-02:
FWIW (as it is closed :) works for me

tiborrr, 2021-03-24:
> "doesn't work for me:"
> 
> Strange. It works on my machine.
> 
> Could you try clearing the cache? rm -rf /home/inaky/.cache/zephyr
> 

For however comes across this thread with the error `xtensa-esp32-elf-gcc: error: unrecognized command line option '-m32'`. The above fixed this too. thanks @SebastianBoe !

PR #7253 (2018-05-02): Fix check compiler flag
Fix a bug where different compiler checks were aliased and therefore
the test results were incorrectly re-used.

The 'check' string is used by CMake internally to cache test results,
but when testing linker flags the check string has been aliased
between the different linker checks. To fix the aliasing issue we add
the CMAKE_REQUIRED_FLAGS variable to the 'check' string. The aliasing
issue disappears because the linker flag-under-test is in
'CMAKE_REQUIRED_FLAGS'.

Also, suppress the compiler check messages because they have become unacceptably verbose.

This fixes #724 [... cut]
commit aa90d721 (2018-05-02): cmake: Introduce a key version to invalidate corrupted caches

When a bug in the caching mechanism is detected the cache format must
be bumped to ensure that we no longer get a cache hit on old corrupted
keys.

This fixes #7246 when the cache is corrupted.