zephyrproject-rtos/zephyr #7249: Arduino 101 / ARC:  tests/kernel/fifo/fifo_api/ fails after 3d9ba10b5c903265d870a9f24065340d93e7d465

inakypg, 2018-04-29:
Starting said commit:
```
    arch: arc: bug fixes and optimization in exception handling
    
    The original exception handling has space to optimize and
    and some bugs need to be fixed.
    
    * define NANO_ESF
       * add the definition of NANO_ESF which is an irq_stack_frame
       * add the corresponding codes in exception entry and handler
       * remove _default_esf
    * implement the _ARCH_EXCEPT
       * use trap exception to raise exception by kernel
       * add corresponding trap exception entry
       * add _do_kernel_oops to handle the exception raised by
         _ARCH_EXCEPT.
    * add the thread context switch in exception return
       * case: kernel oops may raise thread context switch
       * case: some tests will re-implement SysFatalHandler to raise
         thread context switch.
       * as the exception and isr are handled in kernel isr stack, so
         the thread context switch must be in the return of exception/isr
         , and the exception handler must return, should not be decorated
         with FUNC_NORETURN
    * for arc, _is_in_isr should consider the case of exception
    
    Signed-off-by: Wayne Ren <wei.ren@synopsys.com>
```

running on Arduino 101's ARC (and suspected some other ARC platforms and testcases) fail with:

```
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.11.0-875-gd9cdf7bf30 (delayed boot 1000ms) *****
Running test suite fifo_api
===================================================================
starting test - test_fifo_thread2thread
PASS - test_fifo_thread2thread.
===================================================================
starting test - test_fifo_thread2isr
Exception vector: 0x3, cause code: 0x6, parameter 0x0
Address 0x1cf8c0f0
Current thread ID = 0xa8000588
Faulting instruction address = 0x1cf8c0f0
Fatal fault in thread 0xa8000588! Aborting.
Exception vector: 0x3, cause code: 0x6, parameter 0x0
Address 0x1cf8c0f0
Current thread ID = 0xa8000588
Faulting instruction address = 0x1cf8c0f0
Fatal fault in thread 0xa8000588! Aborting.
Exception vector: 0x3, cause code: 0x6, parameter 0x0
Address 0x1cf8c0f0
Current thread ID = 0xa8000588
Faulting instruction address = 0x1cf8c0f0
Fatal fault in thread 0xa8000588! Aborting.
Exception vector: 0x3, cause code: 0x6, parameter 0x0
Address 0x1cf8c0f0
Current thread ID = 0xa8000588
Faulting instruction address = 0x1cf8c0f0
Fatal fault in thread 0xa8000588! Aborting.
Exception vector: 0x3, cause code: 0x6, parameter 0x0
Address 0x1cf8c0f0
Current thread ID = 0xa8000588
Faulting instruction address = 0x1cf8c0f0
Fatal fault in thread 0xa8000588! Aborting.
Exception vector: 0x3, cause code: 0x6, parameter 0x0
Address 0x1cf8c0f0
Current thread ID = 0xa8000588
Faulting instruction address = 0x1cf8c0f0
Fatal fault in thread 0xa8000588! Aborting.
Exception vector: 0x3, cause code: 0x6, parameter 0x0
Address 0x1cf8c0f0
Current thread ID = 0xa8000588
Faulting instruction address = 0x1cf8c0f0
Fatal fault in thread 0xa8000588! Aborting.
Exception vector: 0x3, cause code: 0x6, parameter 0x0
Address 0x1cf8c0f0
...
```
the exception loop repeats for ever, the board is stuck there.

inakypg, 2018-04-30:
Also affects tests/kernel/mem_pool/mem_pool_api:

```
Running test suite mpool_api
===================================================================
starting test - test_mpool_alloc_free_thread
PASS - test_mpool_alloc_free_thread.
===================================================================
starting test - test_mpool_alloc_free_isr
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.11.0-725-gf10fddeb56 (delayed boot 1000ms) *****
Running test suite mpool_api
===================================================================
starting test - test_mpool_alloc_free_thread
PASS - test_mpool_alloc_free_thread.
===================================================================
starting test - test_mpool_alloc_free_isr
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.11.0-725-gf10fddeb56 (delayed boot 1000ms) *****
Running test suite mpool_api
===================================================================
starting test - test_mpool_alloc_free_thread
PASS - test_mpool_alloc_free_thread.
===================================================================
starting test - test_mpool_alloc_free_isr
***** delaying boot 1000ms (per build configuration) *****
***** Booting Zephyr OS v1.11.0-725-gf10fddeb56 (delayed boot 1000ms) *****
Running test suite mpool_api
===================================================================
...
```

nashif, 2018-05-02:
fixed.

closed 2018-05-02 by nashif