zephyrproject-rtos/zephyr #7266: Kconfig syntax error in arch/Kconfig:114

SebastianBoe, 2018-04-30:
There is a syntax error in arch/Kconfig:114

It is not reported by kconfiglib.py, e.g. when running make/ninja, but it is reported by the Kconfig GUI tools like xconfig/menuconfig (that rely on config.c).

SebastianBoe, 2018-04-30:
```
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 109) config DYNAMIC_OBJECTS
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 110) 	bool "Allow kernel objects to be requested on system heap"
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 111) 	default n
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 112) 	depends on USERSPACE
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 113) 	depends on HEAP_MEM_POOL_SIZE > 0
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 114) 	help
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 115) 	Enabling this option allows for kernel objects to be requested from
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 116) 	the system heap, at a cost in performance and additional memory.
31bdfc014e arch/Kconfig          (Andrew Boie             2017-11-08 16:38:03 -0800 117) 

```

Might be due to the help text not being indented. It is not clear why this would trigger now though.

cvinayak, 2018-04-30:
it is the "depends on HEAP_MEM_POOL_SIZE > 0"  I think comparison operation is not supported.

SebastianBoe, 2018-04-30:
Likely culprit, 6 days old commit: https://github.com/zephyrproject-rtos/zephyr/commit/31bdfc014e786d94608b4405694520f41c5066bf

nashif, 2018-04-30:
Isnt this a bug in kconfiglib allowing? it should have have detected the syntax error.

andrewboie, 2018-04-30:
> HEAP_MEM_POOL_SIZE > 0

I could have sworn I copied this from a different Kconfig. I'll work on a fix now.

ulfalizer, 2018-04-30:
@andrewboie 
Kconfiglib supports it. It's the version of the C tools used in Zephyr that doesn't.