zephyrproject-rtos/zephyr #7272: Be more consistent with Kconfig language

SebastianBoe, 2018-04-30:
The language used in Kconfig prompts is inconsistent. This presents application developers a messy and potentially confusing user interface (when the variation in language actually intends to communicate a variation).

Examples of phrasing used to communicate that a driver should be enabled in "Device Drivers":

```
Device Drivers:
  <driver>
  <driver> driver
  <driver> drivers
  <driver> Drivers
  <driver> drivers options
  <driver> support
  <driver> Support
  <driver> Configuration
  <driver> Device Drivers
  Enable <driver> driver
  add support for <driver> Drivers
```

Definition of Done:
[ ] Port the Kconfig language used to be more consistent.
[ ] Develop guidelines that can be used in code-review (and during porting) to enforce consistency.

harichillin, 2026-09-12:
Hi @tejlmand @nashif, following the merged style guide in #95132 and the reset/dac examples, I'd like to pick up drivers/pwm as my first contribution. Let me know if it's free

harichillin, 2026-09-12:
Opened #118901 with the changes for drivers/pwm.

richardmaina3, 2026-09-20:
Hi @tejlmand, following the pattern from the merged reset, dac, and pwm PRs, I would like pick up drivers/can (~29 files) for my first contribution.

minh-anderson-dojofive, 2026-09-28:
Hi @tejlmand @nashif, I would like to pick drivers/eeprom for my first contribution. I've checked that there is no opened PR for it. I'll follow the merged #95132 and #118901