zephyrproject-rtos/zephyr #7309: minnowboard DTS is not configured correctly

andrewboie, 2018-05-02:
According to marketing materials, minnowboard has 2GB of RAM and 4MB of flash.

However, our minnowboard.dts has this:

```
#define DT_FLASH_SIZE		__SIZE_K(72)

#if XIP
	#define DT_SRAM_SIZE		__SIZE_K(128)
#else
	#define DT_SRAM_SIZE		__SIZE_K(256)
#endif
```

Two problems:
1. These values are way too small.
2. The board either uses XIP or it doesn't. AFAIK minnowboard has a bootloader so it doesn't. We should NOT have ifdef XIP here.

andrewboie, 2018-05-02:
This is even worse than I previously suspected -- the build is not using the DT_SRAM_SIZE definition in DTS anyway, it's still getting the data from somewhere else. Was this ever tested??

andrewboie, 2018-05-02:
OK this needs major cleanup.
The RAM size is currently coming from arch/x86/soc/atom/Kconfig.defconfig via CONFIG_RAM_SIZE and CONFIG_ROM_SIZE.
For ATOM the RAM/ROM size needs to be expressed on a per BOARD basis, and derived from DTS.

SavinayDharmappa, 2018-05-03:
RAM ,ROM and XIP data are  derived from arch/x86/soc/atom/Kconfig.defconfig.

nashif, 2018-05-09:
@SavinayDharmappa ping, are you looking into this?

SavinayDharmappa, 2018-05-10:
yeah. None of x86 boards derive RAM and ROM size from dts. So I am moving all boards boards to derive RAM and ROM size from dts. Will push the patch by EOD

SavinayDharmappa, 2018-05-10:
https://github.com/zephyrproject-rtos/zephyr/pull/7451/commits/467983ab9b4051825083dfb77e8bb8188dc08afa

PR #7451 (2018-05-14): dts: x86: derive RAM and ROM size from dts instead of Kconfig
Fixes #7309
closed 2018-05-14 by nashif