zephyrproject-rtos/zephyr #7314: Generate SPDX TagValue document as part of 1.12 release

kestewart, 2018-05-02:
Background:  We're looking at setting up Hyperledger nodes  between LF, Windriver, Intel to track software release and builds and would like Zephyr to be part of it from the start.  This is a way of tracking software provenance, licensing and security related information in the software supply chain. For use in safety critical applications, accurate provenance tracing is going to be extremely important, for adoption.  Any other organizations who wants to participate in hosting a hyperledger node, please let @kestewart know and we can pull you in.     

Two open source options available today to generate the SPDX files are FOSSology & ScanCode.   FOSSology is useful for overall audit and making conclusions based on manual inspection.  ScanCode is more friendly to command line automation, and incorporating into each build.

kestewart, 2018-05-03:
@MaureenHelm - let me know if there is other information you'd like to see added to this request.

nashif, 2018-09-10:
@kestewart please confirm if the generated files are what we need:

http://docs.zephyrproject.org/spdx/zephyr-1.13.0-spdx.tv
http://docs.zephyrproject.org/spdx/zephyr-1.13.0-spdx.rdf

closed 2018-09-10 by nashif

kestewart, 2018-09-10:
Hi Anas,
    Those files don't have effective conclusions to make them useful.

PackageName: zephyr
PackageDownloadLocation: NOASSERTION
PackageVerificationCode: 34e5aefef7e94ef0de4a8fd8643dac7170355029
PackageLicenseDeclared: NOASSERTION
PackageLicenseConcluded: NOASSERTION

We do have a declared license for the package.   The license should be concluded.

nashif, 2018-09-11:
> We do have a declared license for the package. The license should be concluded.

not sure how to tell scancode the declared license for the package.