zephyrproject-rtos/zephyr #7316: native_posix supersedes ztest mocking

andrewboie, 2018-05-02:
We should remove ztest mocking from the tree as we have a much more robust native solution now.

nashif, 2018-05-15:
ztest with native_posix is what meant here. The mocking interface has been duplicating many of the interfaces introducing empty headers and noop functions and it is becoming difficult to manage. Using ztest with native_posix platform gives us the native executable for unit testing without having to replicate the system. I did not see much users of this, do you have cases in mind where this would be used?

needs some more discussion, this is not really a bug, making it an enhancement.

nashif, 2019-06-19:
Thought about it a bit more, I think those are 2 different things completely. We should support mocking in general, whether ztest is the right place for this or if we should support cmock for example is a different question. native_posix does not replace mocking.

closed 2019-06-19 by nashif