zephyrproject-rtos/zephyr #7317: Add generation of SPDX TagValue documents to each build

kestewart, 2018-05-02:
Background: In order to keep up with the licensing changes, and ensure that no "surprises" happen late in the cycle,  we need to monitor the state of license information in the project on every build.

By following the layout proposed in reuse.software (see: https://reuse.software/practices/2.0/) scripting can be developed to check that only files with approved licenses are automatically accepted.

By building an SPDX document on each build,  checks can be programmed, to enforce /ext for deviations from Apache-2.0,  and that all files have a detectable license.   By doing these checks on every build,  licensing issues can be resolved as soon as detected, and reverted/fixed, rather than waiting for audit at formal release.

kestewart, 2018-05-02:
First step is working through https://github.com/zephyrproject-rtos/zephyr/issues/7314

nashif, 2018-09-10:
This is a CI task that is not related to a specific release.

swinslow, 2020-12-28:
Hi @nashif, @kestewart suggested I comment here about this. I've been working on a proof-of-concept to automatically generate tag-value SPDX documents for CMake-based projects.

A prototype is at https://github.com/swinslow/cmake-spdx, with the example using a build of Zephyr's basic/blinky sample project. It leverages the [cmake file-based APIs](https://cmake.org/cmake/help/latest/manual/cmake-file-api.7.html) to create an SPDX 2.2 tag-value document for the sources, as well as a second SPDX document for the built artifacts, including SPDX Relationships to show which built files are generated from which source files.

There's certainly room for improvement, but it might be useful as a starting point for this issue. I'd be happy to get on a call with you and/or others from the team, and walk through how it works if it would be helpful.

closed 2021-05-05 by nashif