zephyrproject-rtos/zephyr #7325: Doc: gather installation instructions for flash/debug tools somewhere

DWagenk, 2018-05-03:
Hi there,

I tried moving over to using Windows natively for zephyr development and stumbled over missing documentation on how to set up openocd for use with zephyr on windows. Since there's more to this topic (how to correctly set up pyocd, nrftool etc.) I'd suggest gathering them in a new section somewhere closely related to the platform dependent getting started guides.

If I'd not have been familiar with using zephyr on linux already, it just not working would have probably turned me away from using zephyr.

I had to explore some of the scripts and finally the right `cmake/flash/CMakeLists.txt` file to know, wich environment variables to set for configuring openocd. I guess the situation is similar for setting up some of the remaining flashing- and debugging-tools.

MaureenHelm, 2018-05-04:
I've done some of this for jlink and pyocd in `doc/tools/opensda.rst`, but that only covers tools used on nxp boards. @mbolivar, @dbkinder what do you think?

mbolivar, 2018-05-04:
I like the idea. What about an umbrella index .rst page for flashing and debugging tools, with a toctree directive inside that indexes pages for each tool?