zephyrproject-rtos/zephyr #7331: Precise time sync through BLE mesh.

hgijeon, 2018-05-03:
Timer of the devices can be shifted as times goes by.
It would be great if IoT devices are automatically synchronizing time from each other.
Is it possible to keep sub-millisecond precision sync between the devices?

raveslave, 2018-11-11:
I'd like to get in touch with someone that's got experience implementing similar things that could be up for making this happen.   

Except for the sync itself, I believe we need an autonomous handover to ensure stability even if the device acting as master leaves. _ref: IEEE 1588 BMC (best master clock)_

carlescufi, 2019-08-15:
CC @pabigot

sslupsky, 2020-03-09:
Could this topic be expanded to encompass time synchronization using methods other than BLE?  For instance, NTP, GPS, LoRa, 802.15.4e, RTC, etc?

Also, I just determined that there is no zephyr api to set the system clock.  So, for the use case where you boot your device and want to manually set the clock to the local time (or some other arbitrary time), there doesn't appear to be an API for this?

closed 2020-10-06 by github-actions[bot]