zephyrproject-rtos/zephyr #7336: USB DFU: this area can not be overwritten

zhaozhongchen, 2018-05-04:
I am tring download a sign-hello.bin by using DFU:

```
sudo dfu-util --alt 1 --device 2fe3:0100  --download signed-hello.bin
dfu-util 0.8

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2014 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to dfu-util@lists.gnumonks.org

dfu-util: Invalid DFU suffix signature
dfu-util: A valid DFU suffix will be required in a future dfu-util release!!!
Opening DFU capable USB device...
ID 2fe3:0100
Run-time device DFU version 0110
Claiming USB DFU Runtime Interface...
Determining device status: state = appIDLE, status = 0
Device really in Runtime Mode, send DFU detach request...
Resetting USB...
Opening DFU USB Device...
Claiming USB DFU Interface...
Setting Alternate Setting #1 ...
Determining device status: state = dfuIDLE, status = 0
dfuIDLE, continuing
DFU mode device DFU version 0110
Device returned transfer size 64
Copying data from PC to DFU device
Download        [                         ]   0%            0 bytes failed!
state(10) = dfuERROR, status(3) = Device is unable to write memory
```

next is the log printed from zephyr's console uart: 

```
***** Booting Zephyr OS v1.11.0-955-g9a265fc *****                                    
[general] [DBG] dfu_status_cb: USB device reset detected, state 0                     
[general] [DBG] dfu_status_cb: USB device supended                                    
[general] [DBG] dfu_status_cb: USB device reset detected, state 0                     
[general] [DBG] dfu_status_cb: USB device reset detected, state 0                     
[general] [DBG] dfu_status_cb: USB device configured                                  
[general] [ERR] dfu_custom_handle_req: DFU alternate setting 0                        
[general] [DBG] dfu_class_handle_req: DFU_GETSTATUS: status 0, state 0                
[general] [DBG] dfu_class_handle_req: DFU_DETACH timeout 1000, state 0                
[general] [DBG] dfu_status_cb: USB device reset detected, state 1                     
[general] [DBG] dfu_status_cb: USB device reset detected, state 2                     
[general] [DBG] dfu_status_cb: USB device reset detected, state 2                     
[general] [DBG] dfu_status_cb: USB device reset detected, state 2                     
[general] [DBG] dfu_status_cb: USB device configured                                  
[general] [ERR] dfu_custom_handle_req: DFU alternate setting 1                        
[general] [DBG] dfu_class_handle_req: DFU_GETSTATUS: status 0, state 2                
[general] [DBG] dfu_class_handle_req: DFU_DNLOAD block 0, len 64, state 2             
[general] [DBG] dfu_class_handle_req: DFU_DNLOAD start                                
[general] [ERR] dfu_class_handle_req: This area can not be overwritten                
[general] [DBG] dfu_class_handle_req: DFU_GETSTATUS: status 3, state 10               
[general] [ERR] dfu_custom_handle_req: DFU alternate setting 0               
```

seems there forget add CONFIG_FLASH_BASE_ADDRESS  in dfu_data.flash_addr.

```
switch (dfu_data.state) {
		case dfuIDLE:
			SYS_LOG_DBG("DFU_DNLOAD start");
			dfu_reset_counters();
			if (dfu_data.flash_addr != FLASH_AREA_IMAGE_1_OFFSET) {
				dfu_data.status = errWRITE;
				dfu_data.state = dfuERROR;
				SYS_LOG_ERR("This area can not be overwritten");
				break;
			}
```

jfischer-phytec-iot, 2018-05-04:
@zhaozhongchen I had CONFIG_FLASH_BASE_ADDRESS there but I removed it and I can not remember why. With which board or SoC did you try it?

zhaozhongchen, 2018-05-05:
@jfischer-phytec-iot stm32f401xe board. maybe you tested with a board that flash address maped to zero?

PR #7337 (2018-05-10): usb: dfu: fix 'this area can not be overwritten'
fixes #7336
closed 2018-05-10 by nashif