zephyrproject-rtos/zephyr #7340: DISCUSS: If condition judgment

zhaozhongchen, 2018-05-04:
some **if condition** in usb_handle_control_transfer is very hard to understand, can we rewrite it?
such as:

```
if (!(setup->wLength == 0) &&
    !(REQTYPE_GET_DIR(setup->bmRequestType) ==
    REQTYPE_DIR_TO_HOST)) {
			return;
}
```

convent to:

```
if ((setup->wLength > 0) && 
    (REQTYPE_GET_DIR(setup->bmRequestType) ==
    REQTYPE_DIR_TO_DEVICE)) {
        return ;
}
```

zhaozhongchen, 2018-05-04:
@jfischer-phytec-iot @ydamigos what's you idea?

finikorg, 2018-05-04:
Maybe have it like
```
if (setup->wLength && REQTYPE_GET_DIR(setup->bmRequestType) == REQTYPE_DIR_TO_DEVICE) {
        return ;
}
```

zhaozhongchen, 2018-05-05:
ok i will make a pr to rewrite some if conditions