zephyrproject-rtos/zephyr #7347: Zephyr builds are slow on Windows

SebastianBoe, 2018-05-04:
The CMake configure-time (time cmake ..) is approximately 6 times slower on Windows than on Linux.

It should first be understood why Zephyr builds slower on Windows, and then depending on what is discovered we should make the performance comparable, e.g. half as fast as Linux would be acceptable.

Benchmarking and analysis:

We need both benchmarking and analysis to be able to resolve this issue. Benchmarking should
include relevant platform information (CPU, OS version, git revision, VM tech if not native).

SebastianBoe, 2018-05-04:
Benchmarking done so far ...

Windows 7, VMWare, git ?, i7-4770: about 3 seconds
Ubuntu 16, native, 542f10de7, i7-4770: 0.6 seconds.

hakonfam, 2018-05-18:
Git: 198b358
HW: i7-7700@3.6GHz
Command ~= `time cmake -H$ZEPHYR_BASE/samples/hello_world -GNinja -DBOARD=nrf52_pca10040 -Bbuild`
Win10: 3.4 seconds
Arch (4.16.3-1) (under Virtualbox): 1 second

SebastianBoe, 2018-05-18:
This implies that it is not related to VM-tech.

It would be interesting to know what parts of Configure run slower on Windows.

E.g. is it processing of CMakeLists.txt code or is it a particular external program that is executed by CMake.

SebastianBoe, 2018-05-18:
What I am hoping for, is that the slowdown is isolated to one or two external programs that run poorly on Windows, but can be easily modified to run as fast as Linux.

carlescufi, 2018-10-04:
I think we can close this now @SebastianBoe, as this is no longer the case.

SebastianBoe, 2018-10-08:
Really? That is good news, I would appreciate it if someone could post a benchmarking result to confirm this.

`time cmake -H$ZEPHYR_BASE/samples/hello_world -GNinja -DBOARD=nrf52_pca10040 -Bbuild`

carlescufi, 2018-10-08:
Oh sorry, I thought you meant the total time. The configure time is still around the same for your example (3.4s) but the total times are now comparable on my machine (21s vs 24s).

GeorgeCGV, 2023-03-06:
> nrf52_pca10040 

Doesn't seem to exist anymore.

Performed for `nucleo_h743zi `:

`time cmake -H$ZEPHYR_BASE/samples/hello_world -GNinja -DBOARD=nucleo_h743zi -Bbuild`

Outcome:

```
real    0m46.832s
user    0m1.250s
sys     0m1.327s
```

- Zephyr 577c678df603798a32aad50918f7683e98eb7f0e
- CMake 3.24.2
- Executed on SSD, Windows 10 22H2

nordicjm, 2024-11-19:
Windows is a different OS with a different kernel, it also has an inbuilt virus scanner enabled globally which is well known to cause compilation to be slow, thus it is not right to compare them equally and it is expected that windows will be slower to compile on the same hardware than linux. If this is still an issue and it can be shown that the difference is big when the host windows OS has been configured to e.g. not have antivirus running then it can be re-opened

closed 2024-11-19 by nordicjm