zephyrproject-rtos/zephyr #7348: subsys/fs/: shell command for mounting the file system is unimplemneted

nvlsianpu, 2018-05-04:
After introduction of VFS the mount operation is required in order to setup the FS.

ramakrishnapallala, 2018-05-09:
@nvlsianpu what is expected input here? are you expecting something like below?

mount NFFS "nffs"

koffes, 2018-05-22:
Yes, this would be beneficial. If appropriate, the file system could also be mounted automatically when the relevant settings have been selected. Alternatively have a separate setting for auto mount.

nvlsianpu, 2018-05-23:
@ramakrishnapallala 
Yes - something like that.
We simultaneously might looks forward for automatically mounting the FS.

ramakrishnapallala, 2018-05-28:
moving to v1.13

varun-sha, 2018-06-28:
fs Shell module is created, which can be enabled by CONFIG_FS_SHELL=y
mount commands example:-
format:- fs mount <fs e.g: fat/nffs> <mount-point>
example commands:- 
fs mount fat /RAM
fs mount nffs /nffs

varun-sha, 2018-07-12:
@nvlsianpu @nashif @ramakrishnapallala 
pls review https://github.com/zephyrproject-rtos/zephyr/pull/8384

ramakrishnapallala, 2018-07-27:
@varun-sha I think you need to migrate the PR to new shell implementation from Nordic.

varun-sha, 2018-08-06:
@ramakrishnapallala i think Nordic cli is not targeted for 1.13 release
@nvlsianpu @koffes : can you pls comment, i have done implementation for this and submitted pr as per current cli which we have in zephyr, should we go ahead with this pr? or should wait for nordic cli which will be integrated into Zephyr..and then integrate our changes into that...

koffes, 2018-08-06:
I have no say here, but if we can have this feature as part of the existing CLI instead of waiting it would be beneficial.

varun-sha, 2018-08-13:
@nashif  Can you share your thoughts...

PR #8384 (2018-09-13): Introduce shell for File system mount operation
subsys: shell : fs new shell module for fs mount operation
Fixes #7348, also includes test cases
closed 2018-09-13 by nashif