zephyrproject-rtos/zephyr #7375: Codecov does not report coverage regressions from untested code

SebastianBoe, 2018-05-07:
PR https://codecov.io/gh/zephyrproject-rtos/zephyr/pull/6748/

demonstrates that codecov is not able to detect when new code that is untested is committed. Such code should be considered a coverage-regression. It is not clear why it does not catch this.

aescolar-ot, 2018-05-18:
As of now we are only feeding codecov the code that is compiled and run by sanitycheck in the native_posix tests.

SebastianBoe, 2018-05-18:
Right, that would explain it. I guess this will be difficult to resolve then.

aescolar-ot, 2018-05-18:
Yep, it would not be easy to resolve. Our only source of coverage are the native_posix runs. So code in, for ex., arch/arm is out of scope for the current setup. And so are all boards and drivers.
You can see here what is actually part of coverage analysis:
https://codecov.io/gh/zephyrproject-rtos/zephyr/tree/master

nashif, 2019-11-21:
we are not using codecov anymore, so closing.

closed 2019-11-21 by nashif