zephyrproject-rtos/zephyr #7384: ARM MPU region configuration possibly out of bounds

andrewboie, 2018-05-07:
in arm_mpu.c arm_core_mpu_configure_mem_partition(), the region index is computed from _get_region_index_by_type(THREAD_DOMAIN_PARTITION_REGION).

THREAD_DOMAIN_PARITITON_REGION is defined as 4.
However, consider a SOC like the Beetle. mpu_config.num_regions = 6. _get_region_by_type(THREAD_DOMAIN_PARITITON_REGION) will return 9, which is greater than the maximum amount of MPU regions (0-7)

At minimum needs some assertions here.

ioannisg, 2018-05-29:
@andrewboie I believe you're suggesting ASSERTS in the arm_mpu.c driver, so certain operations will return error (and be caught by other parts of the code)?

ioannisg, 2018-05-31:
Could this be closed, @andrewboie?

linkmeyer, 2018-06-05:
@andrewboie , did you see the question above about whether this bug can be closed?

PR #7902 (2018-06-27): arm: clean up MPU code for ARM and NXP
* We are now *much* better at not reserving unnecessary
system MPU regions based on configuration. The #defines
for intent are now an enumerated type. As a bonus, the
implementation of _get_region_index_by_type() is much
simpler. Previously we were wasting regions for stack guard
and application memory if they were not configured.

* Certain parts of the MPU code are now properly ifdef'd
based on configuration.

* THREAD_STACK_REGION and THREAD_STACK_USER_REGION was a
confusing construction and has now been replaced with
just THREAD_STACK_REGION, which re [... cut]
commit 8bcffefb (2018-06-27): arch: arm: clean up MPU code for ARM and NXP

* We are now *much* better at not reserving unnecessary
system MPU regions based on configuration. The #defines
for intent are now an enumerated type. As a bonus, the
implementation of _get_region_index_by_type() is much
simpler. Previously we were wasting regions for stack guard
and application memory if they were not configured.

* NXP MPU doesn't reserve the last region if HW stack
protection isn't enabled.

* Certain parts of the MPU code are now properly ifdef'd
based on configuration.

* THREAD_STACK_REGION and THREAD_STACK_USER_REGION was a
 [... cut]
closed 2018-06-27 by andrewboie