zephyrproject-rtos/zephyr #7389: t1_adc108s102 buffer overflow due to chan->buf_idx growth

andrewboie, 2018-05-07:
Sampling is done in an infinite loop until no more data is received from SPI for that transaction (_ti_adc108s102_sampling() returns 0).

For each iteration, _ti_adc108s102_handle_result() is called in increments chan->buf_idx +=2 for each entry.

The check for free space in the buffer (entry->buffer_length - chan->buf_idx == 0) only works if entry->buffer_length is an even number. chan->buf_idx will keep increasing and writing out of bounds *((u16_t *)(entry->buffer + chan->buf_idx)) = ADC102S102_RESULT(adc->sampling_buffer[s_i]);

nashif, 2018-05-16:
@tbursztyka can you take a look at this please?

tbursztyka, 2018-05-16:
indeed, buffer's length being an even number should be checked and if not even: that's an error.

PR #7573 (2018-05-17): drivers/adc: Uneven buffers will lead to buffer overflow
TI's ADC108S102 is a sampling on 16 bits, and thus requires the
destination buffer to be made of an even number of bytes.

Fixes #7389
commit cd580e0b (2018-05-17): drivers/adc: Uneven buffers will lead to buffer overflow

TI's ADC108S102 is a sampling on 16 bits, and thus requires the
destination buffer to be made of an even number of bytes.

Fixes #7389
closed 2018-05-17 by nashif