zephyrproject-rtos/zephyr #7390: pinmux subsystem API is undocumented and does not enforce validation

andrewboie, 2018-05-07:
None of the pinmux APIs in include/pinmux.h have any documentation.
In addition, if we are to expose pinmux APIs to user mode, we are going to need the specification to enforce that drivers validate the parameters passed in. Currently the pinmux drivers don't validate any inputs, and a user thread using these APIs have been demonstrated writing to memory that it should not have access to or even disable the MPU.

Until this is done, the pinmux subsystem APIs are being removed from user mode access.

MaureenHelm, 2018-05-10:
@andrewboie Will you please rewrite this such that it describes what needs to be done, rather than just describing the current state of affairs? Something like "Add documentation to the pinmux APIs" and "Validate pinmux API inputs". I'd like to tag this issue as an enhancement, but that doesn't make sense with the current description.

nashif, 2019-02-19:
pinmux system will be completely overhauled and replaced, so closing this old issue.

closed 2019-02-19 by nashif