zephyrproject-rtos/zephyr #7403: RAM flash driver emulator

nvlsianpu, 2018-05-08:
Such  driver will be useful for testing modules which extensively use the flash driver. It will be possible to emulate flash driver functionality for qemu as well.
the driver configuration should allow to configure:
- erase block size
- write block size
- size of emulated flash
- absolute offset of emulated flash

and optionally:
- erase time duration
- write time duration

Something which can be a prototype for such a driver is implemented for NFFS tests (e.g. tests/subsys/fs/nffs_fs_api/src/test_ram_backend.c)

nvlsianpu, 2018-05-08:
@carlescufi @pizi-nordic

pizi-nordic, 2018-05-08:
I would add some features allowing to test rainy day scenarios:
- Ability to generate random failures of read/write/erase operations at given probability (such feature will allow to easily check error handling in subsystems built on top of flash storage).
- Ability to emulate typical flash failures (stuck bit, random inversion on read etc). This feature will be important for devices with expected long operation or when NAND flash will be used for a storage.

nashif, 2019-05-22:
already implemented

closed 2019-05-22 by nashif